AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #505: Potential menubar icon high CPU usage bug

Iazos, 2021-12-21:
**Describe the bug**
Displaying a Menubar Icon has much higher CPU usage than without. Another side effect is Al Dente can never go into the background with the Menubar Icon displayed.

**Screenshots**
[image]

**MacBook**
- Model: Air 2020 M1 (512GB storage, 8 GPU cores)
- macOS version: 12.1
- AlDente Pro, version 1.12

MatthiasKerbl, 2022-01-15:
Hi @nermur,

Do you mean the menubar icon or the dock icon?

MatthiasKerbl, 2022-03-02:
Hi @nermur,

Do you still encounter this problem with our newest version 1.14?

closed 2022-03-07 by MatthiasKerbl