AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #506: Can't install app. Always ask to install helper.

codedeveloper2020, 2021-12-22:
**Describe the bug**
After first lunch app require install helper. After it installed, app not full started and show install helper again. And repeat after trying to do so.

**To Reproduce**
Steps to reproduce the behavior:
1. Lunch app
2. Show popap "AlDente-Pro is trying to install a new helper tool". Enter password and click "Install Helper" button.
3. Click app icon in toolbar. 
4. Open popup message dialog window with two buttons: "Install Helper" and "Quit".
5. When click on button "Install Helper" - nothing is happen.

**Expected behavior**
The app is full start and work.

**Screenshots**

https://user-images.githubusercontent.com/60083023/147130400-96595408-a796-4d88-8aea-80181db5aab5.mov

[image]

**MacBook (please complete the following information):**
 -  MacBook Pro (13-inch, M1, 2020)
 - macOS version 12.1 (21C52)
 - AlDente Pro 
 - AlDente last version

sitnikovdev, 2021-12-23:
The problem was in old version of AlDente Pro. Install new version and it work.

closed 2021-12-23 by sitnikovdev

derekhongg, 2022-01-07:
Where did you find the newest version of Aldente Pro?

MatthiasKerbl, 2022-01-07:
> Where did you find the newest version of Aldente Pro?

Hi @derekhongg,

you can find the newest version here: https://github.com/davidwernhart/AlDente/releases

Cheers,
Matthias

sitnikovdev, 2022-01-07:
Hi,
I wrote in support and they give me new link. Try this one in attachment.