AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #509: Keep getting message to remove old helper

mcartagenah, 2021-12-23:
**Describe the bug**
I updated the app to the latest version and I get a message asking to remove the old helper, I removed it and it keeps appearing and can't do anything else.

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook Pro (16-inch, 2019)
 - Version 12.1 (21C52)
 - AlDente Pro
 - AlDente 1.12

fusuyfusuy, 2021-12-25:
can confirm, having the same problem. 

M1 Macbook Air
Monterey 12.1
Aldente 1.12

MatthiasKerbl, 2021-12-28:
Hi @mcartagenah and @fusuyfusuy,

Please follow the instructions for uninstalling the AlDente manually on this blog post: https://apphousekitchen.com/how-to-uninstall-aldente-pro/

Afterward, install the newest version of AlDente again.
This time, you should get the error message.

Cheers,
Matthias

MatthiasKerbl, 2022-02-03:
Hi @mcartagenah,

I will close this issue for now. If you still have this issue, please let us know.

Cheers,
Matthias

closed 2022-02-03 by MatthiasKerbl

ysomad, 2022-05-10:
Same problem.