AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #514: An option to make AlDente persist throughout sleep, log out, etc.

drukmano, 2021-12-25:
**Is your feature request related to a problem? Please describe.**
Having a problem while switching users and my Mac is starting to charge.

**Describe the solution you'd like**
I understand M1 Macs are resetting the configuration while off, but it would be nice to have a "Persist" button for AlDente to set a fixed limit to the system configuration. This would allow switching users, and even closing AlDente, while still preserving the Mac in a non-charging state.

**Additional context**
Using MacBook Pro 14" M1 Pro and AlDente Pro.

MatthiasKerbl, 2021-12-27:
Hi @drukmano,

Since AlDente Pro version 1.10 we have implemented functionality that supports switching of users. I will write a blog article about this soon to clarify some things about this.

Regarding the Charge Limiter working during sleep or app closed, you can read the following blog articles about how AlDente Pro handles these scenarios.
https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/
https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/

I know that having a charge limit that persists in every scenario would be ideal and is actually available for intel MacBooks through Intel Mode, however, due to many low-level changes of the new Apple Silicon MacBooks, we were not able so far to find such a pathway for the newer MacBooks.

Cheers,
Matthias

closed 2021-12-27 by MatthiasKerbl

drukmano, 2021-12-28:
Hi @MatthiasKerbl 

Since AlDente is capable of limiting the charge when the Mac is in sleep mode, I assumed that it would also be capable of doing it when closing the app (or switching users, for example), using that same mechanism.
What is the difference then? If AlDente is setting a system configuration to disable charging, then why does it still need to be in an active state?

Thanks,
Ohad.

MatthiasKerbl, 2021-12-29:
Hi @drukmano,

Unfortunately, it is a little bit more complicated.

We explained as much as possible what AlDente Pro can do in the articles on our blog:
https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/
https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/

Cheers,
Matthias