AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #515: Charge Limit works only if set over 50%

jakuSk, 2021-12-27:
**Describe the bug**
If Charge Limit is set under 50% (like 40%), device will charge to 50-52% no matter what is set as Charge Limit. If Charge limit is set over 50% Charge limit works as it should.

Description with "Menubar icon status symbol" 
I set AlDente Charge limit to 40%. Click on the Discharge button. MacBook discharges to 40% like always. Then Menubar icon does little dance 3-5 times. Icon goes from pause, to charing, then to pause again. after 3-5 times goes to changing and charges to 50-52%.

**To Reproduce**
Steps to reproduce the behavior:
Set Charge Limit under 50%
let the device Discharge (unplugged cable/with button in AlDente)

[image]

MacBook Pro (13-inch, 2020, Four Thunderbolt 3 ports) - MacBookPro16,2
 - macOS version Monterey 12, 12.1
 - AlDente Pro
 - AlDente Version 1.0, 1.1, 1.12
 - List of all enabled/disabled settings in AlDente:

Hi, Unfortunatelly nothing worked for me. Fresh reinstall of MacOS with usbDrive full wipe. Reset SMC, NVRAM. Still behaves the same. Only way to set charge limit under 50% is with intel mode which is for me a bit useless.

karpisdiem, 2022-01-11:
Hi I am having the same exact issue. If I set the charge limit lower than 50% while the laptop lid is closed and connected to external monitors, the discharging will not work. If I keep spamming the discharge button to anything lower then 50% my monitors will disconnect, then the laptop will go into sleep mode, after a couple of seconds it will turn everything back on and show that discharging is not possible. However once I set the discharge limit to 50% or higher, it works as it should. I've tried everything as well. Can you let me know if you were able to successfully fix this?

MatthiasKerbl, 2022-01-15:
> 

Hi @jakuSk,

Thank you for this detailed bug report.
We are already aware of this bug and only very few of our users experience it.
Unfortunately, we do not know yet what the cause of this issue is and can therefore not solve it currently.

I will update this thread when we find anything new.

Cheers,
Matthias

madkep, 2022-03-03:
I also have the same problem as mentioned here.
MacBook Pro (13-inch, 2018, Four Thunderbolt 3 ports)
Monterrey 12.2.1
all versions of aldente
Has anyone discovered a solution?

jakuSk, 2022-05-23:
Hi @MatthiasKerbl, are there any news regarding this topic?
thank you for your reply

MatthiasKerbl, 2022-05-23:
> Hi @MatthiasKerbl, are there any news regarding this topic? thank you for your reply

Hi @jakuSk,

Unfortunately, we do not have much news on this topic.
One user reported he had this issue but after a while, it was gone by itself and he told us that he did nothing specifically to fix it.
So far we were also unlucky in finding the cause of this bug.

When I know anything new, I will let you know.

Cheers,
Matthias

MatthiasKerbl, 2022-11-07:
Hi everyone,

Unfortunately, still no good news about this bug. However, we would like to track the MacBook models which have this issue.
So far it was reported on the following models:

- MacBook Pro (13-inch, 2020, Four Thunderbolt 3 ports)
- MacBook Pro (16-inch, 2019)
- MacBook Pro (13-inch, 2018, Four Thunderbolt 3 ports)

Cheers,
Matthias

DamonHD, 2022-11-07:
Would it be useful for me to try this on my MBA M1 Ventura 13.0?  Eg set the limit to 40% (instead of my normal 80%) for a while and see if it works as expected?

Rgds

Damon

MatthiasKerbl, 2022-11-07:
> Would it be useful for me to try this on my MBA M1 Ventura 13.0? Eg set the limit to 40% (instead of my normal 80%) for a while and see if it works as expected?
> 
> Rgds
> 
> Damon

Hi @DamonHD,

This would definitely give us at least one more data point.
Currently, we are trying to find out if it is only happening to certain MacBook models and if it is always or just sometimes happening to those MacBook models.

Cheers,
Matthias

DamonHD, 2022-11-07:
Data point coming up then ... as soon as my first ever calibration run finishes!  So much excitement in one day.  B^>

Rgds

Damon

threadflow, 2022-11-07:
My machine is MacBook Pro (16-inch, 2019).
AlDente works at the very beginning.
But one day, the bug is coming, and it remains till now.

DamonHD, 2022-11-10:
Has behaved well for me over sample of ~3d, see above.

Rgds

Damon

MatthiasKerbl, 2022-11-10:
> Has behaved well for me over sample of ~3d, see above.
> 
> Rgds
> 
> Damon

Thanks for testing. That's good to hear, it would have been the first Apple Silicon MacBook otherwise.

Cheers,
Matthias

benapetr, 2022-12-26:
having this on MBP 2018, had my battery replaced in the past, might be related?

cepresso, 2024-01-18:
Hey, I have a MacBook Pro 16", 2019 and the same issue.
I discovered, that enabling [Intel Mode](https://apphousekitchen.com/feature-explanation-intel-mode/?utm_campaign=aldente-1.24.1&utm_medium=questionmark-icon&utm_source=aldente-app) makes a setting < 50% work correctly.
The downside is that all the other interesting features are not working :-(
Is it maybe an option to add some special logic to use Intel mode to fix the charge limit for that special cases?