AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #516: displayed charge is limited  -- but battery seems 100% charged

randfee, 2021-12-28:
1. Charge limit was set to 85% (AlDente free version 1.12).
2. I charged the MacBook Pro 16" M1Max with the fast-charging power supply via MagSafe. 
3. the battery indicator stopped at 85% (I use MenuMeters but the apple one shows the same) but the laptop remained plugged in
4. I was on and off working on some office documents, email, browsing, music listening. typically the machine lasts for 10h during this type of work when charged to 100%.
5. I unplugged the laptop roughly 2 hours after it hat reached the 85% charge limit - during which I took a break
6. I kept working unplugged and noticed after 5-6h of working, that the battery indicator did not drop a single percent over almost 3 hours and then dropped at a near steady 10% per hour

Since the laptop can't have created those estimated 10 Watt hours of energy out of thin air, I really wonder what is going on.
I was thinking about purchasing a pro license, but this seems very suspicious. 

regards
Arnold

=========================

This screenshot with comments should illustrate it all:
[
[image]
](url)

===============

Model Name:	MacBook Pro
  Model Identifier:	MacBookPro18,2
  Chip:	Apple M1 Max
  Total Number of Cores:	10 (8 performance and 2 efficiency)
  Memory:	64 GB
  System Firmware Version:	7429.61.2
  OS Loader Version:	7429.61.2
  Serial Number (system):	XXXXXXXXXXXXX
 
Mac OS 12.1 (21C52) Monterey)

AlDente 1.12 (free)
Settings: Charge limit set to 85%

MatthiasKerbl, 2021-12-28:
Hi @randfee,

Your MacBooks behavior is definitely not as expected.
I have a 2016 15" MacBook Pro and when my MacBook reaches the charge limit, it definitely stops charging. I can confirm this by the internal readout of the charging state and I also have a wattmeter between my charger and the socket which shows 0 watts.
I think for your MacBook it is just a displaying issue. Since macOS does not show the "real" SoC of your battery for various reasons, it shows another SoC which differs by 5-10% from the real SoC.
If you want to check if your MacBook does really stop charging when your charge limit of 85% is reached, you can click on the menubar on the Apple logo/About this Mac/System Report/Hardware/Power. Here you can see at Charge Information, if your MacBook does really stop charging.

Cheers,
Matthias

randfee, 2021-12-30:
well. the OS reports the battery is done charging, but the power consumption of the power supply remains at 15-20W, measured on the socket. LED of the MagSafe is also green. 
Also, the entire Mackbook remains quite warm to the touch, not hot (then again the M1 Max barely ever gets hot). Maybe this is a different bug, I'll open up a new issue with that.

Busnic519, 2022-01-07:
Hi. Didn't want to create a separate thread.
I'm using AlDente Free on my 2019 16" MBP (Intel).
It worked perfectly until I upgraded to Monterey.
Now it won't keep the battery at desired level. 
Discharge mode works, but as soon as it reaches the set level, it starts to charging up to 100% again. 
Also, while in Discharge mode, it grays out the "Sleep" button. 
When I manage to put it to sleep, it will always get fully charged when I resume working.
I tried removing and reinstalling the app, doesn't help. "Optimized charging" is off.

elulcao, 2022-02-15:
I noticed the same `Manage battery longevity` in Monterey, what is the impact of having this feature on? I couldn't find any reference in documentation about disabling or enabling. 

[image]
[image]
[image]

MatthiasKerbl, 2022-02-15:
> I noticed the same `Manage battery longevity` in Monterey, what is the impact of having this feature on? I couldn't find any reference in documentation about disabling or enabling.
> 
> [image] [image] [image]

Hi @elulcao,

When "Manage battery longevity" is enabled, macOS reduces your maximum capacity to 95%. Therefore, even when you charge it to 100% and macOS shows 100% battery it is actually only charged to 95%.

I personally have "Manage battery longevity" disabled since I limit the charging with AlDente and when I want to charge to 100% I want to use the full capacity of the battery and not only 95% of it.

Cheers,
Matthias

MatthiasKerbl, 2022-03-02:
Hi @randfee,

Do you have any new findings regarding your issue?

closed 2022-03-09 by MatthiasKerbl