AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #518: It does not charge anymore

jeffonmac, 2021-12-28:
**Describe the bug**
no charging after fully decharged. 

**Expected behavior**

After my 15 inch MacBook Pro has been fully discharged. Having enabled disables the charging action when the Mac is turned off or in standby. It does not charge anymore. Do you have a solution?

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - macOS version (https://support.apple.com/en-us/HT201260)
 - MacBook Pro 15 inch 
 - AlDente Pro 
 - AlDente Version : AlDente 1.12
 - List of all enabled/disabled settings in AlDente:

Enabled disables the charging action when the Mac is turned off or in standby.

MatthiasKerbl, 2022-01-15:
Hi @jeffonmac,

Usually, this is due to your MacBook discharging into the battery reserve. Therefore, if you plug in your MacBook again it will very cautiously and slowly start charging again until it is in a normal SoC again. Thus it takes longer until it reboots.
Did your MacBooks reboot after a while again?

PearApple151, 2022-01-17:
> **Describe the bug** no charging after fully decharged.
>
> **Expected behavior**
>
> After my 15 inch MacBook Pro has been fully discharged. Having enabled disables the charging action when the Mac is [...]

hey did your Mac start working again?

jeffonmac, 2022-01-18:
Yes it’s good thank you !

closed 2022-01-18 by jeffonmac