AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #519: Macbook remains on (?) - keeps burning 20W of power after charging (even at 100%)

randfee, 2021-12-30:
even after having set the maximum charge to 100%, the macbook (specs below) remains warm to the touch (lid closed!!!). I've tested this several times now. The LED on the MagSafe plug is emitting green, indicating fully charged. 
The power supply keeps pulling 15-20W from the outlet, measured by well calibrated power meter right at the socket.

Logging tools like menumeters don't show any logged data (temps or energy usage) during the lid-closed time but there is no way to tell why the laptop is so warm. Is it running or burning away the energy trying to charge the battery beyond 100% (which would be super dangerous?!)

Uninstalling the helper app and closing AlDente, then closing the lid again immediately stops the power usage (on the power supply) and consequently leads to the entire laptop cooling off to room temperature and remaining there. 
Opening the lid, starting AlDente and re-installing the helper app, then closing the lid leads to the 15W+ constant power consumption and warm laptop again.

therefore, I deem this a very critical behaviour to figure out and fix!

===============

Model Name: MacBook Pro
Model Identifier: MacBookPro18,2
Chip: Apple M1 Max
Total Number of Cores: 10 (8 performance and 2 efficiency)
Memory: 64 GB
System Firmware Version: 7429.61.2
OS Loader Version: 7429.61.2
Serial Number (system): XXXXXXXXXXXXX

Mac OS 12.1 (21C52) Monterey)

AlDente 1.12 (free)
Settings: Charge limit set to 85%

cmalex23, 2022-01-18:
Do you have the same power draw and warm machine even at 85% charge limit or only at 100%?
I'm using AlDente on an M1 Pro 16 with charge limit at 60-70% and the machine doesn't get warm. I don't have a device to measure power from the outlet but since it doesn't get warm I suppose it doesn't draw more than needed.

mtrifiro, 2022-01-21:
I have noticed something similar. If I close my lid and  sleep my Macbook it is very warm after time. I agree, this seems counter to how this app is supposed to behave and it concerns me. Are we sure this app yields the benefits it promises?

mtrifiro, 2022-01-21:
BTW, I have only tried this with my charge set to 85%. I've not yet tried letting it go to 100% and then sleeping but based on the other comments it seems the result will be the same.

MatthiasKerbl, 2022-01-22:
Hi there,

Thank you for your bug report. We are aware that some users experience this issue and we are currently investigating it. Hopefully, we can fix it soon with an update.

MatthiasKerbl, 2022-02-02:
Hi everyone,

We have just released a new version of AlDente (1.14) which should fix the issues you experienced.
If you still encounter any of the issues after updating, please report back.

Download AlDente 1.14 here (https://github.com/davidwernhart/AlDente/releases) or update within the app.

Thank you for your patience.

Cheers,
Matthias

closed 2022-02-02 by MatthiasKerbl