AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #522: AlDente charge limit killed my battery?

taylor325, 2021-12-31:
Hello! I use AlDente on my M1 MacBook Air.
I use from charger. Charge limit: 50%.
Cycle count: 27
But! Today I watched netflix episode from battery and my battery It dropped from 50 to 35% in 45 minutes.
This is normal??? I think no. May I need another charge limit setting or sometimes I need to drain the battery?
Please help I don't want to kill my macbook battery!

laurazard, 2022-01-02:
From the [README](https://github.com/davidwernhart/AlDente#readme)

> Keeping your battery at a lower percentage, such as 20-70%, over weeks without doing full cycles (100%-0%-100%) can result in a disturbed battery calibration. When this happens, your Macbook might turn off with 40-50% left or your battery capacity will drop significantly. However, this is only due to a disturbed battery calibration and not because of a faulty or degraded battery. To avoid this issue, we recommend doing at least one full cycle (0%-100%) every two weeks. Even if your battery calibration gets disturbed, doing 4-5 full cycles will recalibrate your battery and the capacity will go up again. AlDente Pro has a feature called Calibration Mode(https://apphousekitchen.com/feature-explanation-calibration-mode-2/), which will automatically do a full cycle when startet. You can read more about the calibration of "smart" batteries here: https://batteryuniversity.com/article/bu-603-how-to-calibrate-a-smart-battery

I'd suggest doing a few full cycles to recalibrate your battery.

joujek, 2022-01-12:
not sure how safe this app is... 
I was using it for 3 months - covering all instructions, steps described by developer... finally my battery capacity drops from 100% to 90%.... I have followed the full cycles, calibration etc......
then I uninstall this app, checked battery optimisation and after next 6 months I am still at 90% health... 
I guess it is still experimental and as mentioned in readme "you are using on your own risk... " and I paid the 10% of health/capacity as the risk.

MatthiasKerbl, 2022-01-13:
> not sure how safe this app is... I was using it for 3 months - covering all instructions, steps described by developer... finally my battery capacity drops from 100% to 90%.... I have followed the full cycles, calibration etc...... (35 cycles - did 8 cycles forced as recommended recalibration) then I uninstall this app, checked battery optimisation in settings and after next 6 months I am still at 90% health... I guess it is still experimental and as mentioned in readme "you are using on your own risk... " - I have paid the 10% of health/capacity for the mentioned risk.

Hi @joujek,

Regarding your question about how safe AlDente is I can say that using a charge limit is much better for the longevity of a battery than not. This is shown in many studies even with the same cell chemistries as the ones in MacBooks.
That being said, long-term battery health comes down to so many different parameters (SoC, temperature, cycles, charging/discharging speed, calendar age, manufacturing defects,...), therefore it is impossible to have the perfect outcome for everyone. On top of that comes uncertainty such as measurement inaccuracies. However, with AlDente we try to tackle the most important parameters (SoC, temperature, cycles,...) for long-term battery health and give users the option to do something for their battery health.

When it comes to your certain scenario, it is common and normal to lose approximately 5% of battery capacity in the first couple of months regardless of usage, due to effects in the cell chemistry. However, losing 10% that quickly especially with such a low cycle count is not average but might still be not that far off, especially considering the margin of error.

Anyway, I hope I could give you a clearer picture of the situation and the goal of AlDente.

Cheers,
Matthias

MatthiasKerbl, 2022-01-13:
> Thank you!

Hi @taylor325,

Did you see an improvement again after following the instructions in our readme?

joujek, 2022-01-13:
> Hi @joujek,
>
> Regarding your question about how safe AlDente is I can say that using a charge limit is much better for the longevity of a battery than not. This is shown in many studies even with the [...]

Hi @MatthiasKerbl ,
Thank you for explanation - I know the theory, I familiarised with all materials you are provieding on website, I also have electronic graduation. My case is too old to give you valuable logs etc... 
I mentioned here my experience with the app only - my doubt is related if "low level" ingestion into apple's power management is good at all as the most improvement in the app is less than one year old.
You might be right that my example is limited / not general - keep fingers crossed for other's good experience.

m4rkw, 2022-02-14:
> the mac did not limit it to 80% despite the option

Optimised battery charging doesn't just limit the charge level to 80% constantly, it evaluates when you normally unplug the machine from power and then holds it at 80% until a few hours before that time and then charges the remaining 20%. iPhones do the same thing. I'm not aware of any other reports of this feature simply not working so I'm guessing it's likely that you simply misunderstood how it works?

closed 2022-02-25 by MatthiasKerbl

FreeRacing5, 2022-06-24:
Ive been using aldente free for a while cause i quite litteraly murdered my battery on my mbp, battery health dropped from 82% to 30% on coconut battery, and without a doubt, this works like a dream, battery health is now back to 82% and if i need it topped up before i go somewhere i drag the slider to 100%. Thank you to the people that made this app, you guys are absolute legends!

m4rkw, 2022-06-24:
> Ive been using aldente free for a while cause i quite litteraly murdered my battery on my mbp, battery health dropped from 82% to 30% on coconut battery, and without a doubt, this works like a [...]

All that's changed is the numerical output from an algorithm, apps cannot repair battery chemistry. Likely all you needed was a few full charge/discharge cycles to recalibrate the algorithm.