AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #525: Screen brightness sensor signalling full brightness (although in a dark room)

RahulBhalley, 2022-01-02:
**Describe the bug**

The light sensor on my MacBook Pro (13-inch, 2016, Four Thunderbolt 3 Ports) is saying that the environment is extremely bright. So the Touch Bar is bright to fullest, keyboard backlight is not ON, the screen has more white tone (as it is when the screen is fully bright; this one goes to normal screen tone when I turn OFF Automatically adjust brightness). 

**To Reproduce**
I just installed this software and after few hours this started to happen. I also reinstall the software & even uninstalled it but the issue persisted. So I removed the Macintosh HD using Disk Utility in Recovery Mode and made a fresh install of macOS Monterey 12.1 but still the same behaviour persists. I searched online and it looks like that SMC should be reset. So I tried that too by shutting down > holding left shift, control, option, and power button for ~10 seconds > boot but still not fixed!!! What happened to my MBP. This is a panicking situation. 

**Expected behavior**
There should've been no mess up with my screen, Touch Bar, and keyboard backlight brightness at all. 

**Screenshots**
[image]
[image]
[image]
[image]

**MacBook (please complete the following information):**
 - MacBook model: **MacBook Pro (13-inch, 2016, Four Thunderbolt 3 Ports)**
 - macOS version: **macOS Monterey Version 12.1 (21C52)**
 - AlDente Free
 - AlDente Version: **1.12**
 - List of all enabled/disabled settings in AlDente: 
1. Show on Startup: ON
2. Show Dock Icon: OFF
3. Launch at Login: ON
All other settings are OFF.

**Additional context**
N/A.

RahulBhalley, 2022-01-03:
Come on, someone please help! My MBP is now not even charging sinking down (11% right now) and showing plug sign on battery in menu bar!!!!!!!!!!!!!!!!!!!!!!!!!

RahulBhalley, 2022-01-03:
YOU JUST LOST A STAR. 👎 VERY BAD FOR MESSING UP MAC AND NOT EVEN HELPING NOW.

jakuSk, 2022-01-03:
LOL @RahulBhalley. You posted this Yesterday. People had Holidays, celebrating new year....Besides the Disclaimer is very clear:

Disclaimer:

I do not take any responsibility for any sort of damage in result of using this tool! Alltough this had no negative side effects for me and thousands of others, AlDente still taps in some very low level system functions that are not ment to be tampered with. Use it at your own risk!

Copyright(c) 2021 AppHouseKitchen

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.

RahulBhalley, 2022-01-04:
@jakuSk thanks for replying, no offence, but it wasn't much helpful. Almost every open source software (even big corps.) has such disclaimer. I think I'll have to look into code to fix my machine. The brightness sensor is behaving like I'm sitting under sunlight. I doubt if someone even wanna buy my MBP at this situation. But anyone has any solution please let me know. :)

andrewcrook, 2022-03-19:
Did you solve this issue?

RahulBhalley, 2022-03-22:
Nope. Still stuck.

andrewcrook, 2022-03-22:
> Nope. Still stuck.

I dont know macOS under the hood that well. 
I haven’t experienced any issues, however, I am on Apple Silicon which has some differences at the hardware level.
I have had a think and come with some pointers. 

So you correctly reset SMC ?
https://support.apple.com/en-gb/HT201295

Did you change the AlDente settings back before uninstalling?
> To uninstall AlDente, you just have to set the charging limit back to 100%, wait 30 seconds for changes to take place, quit the app and delete the AlDente program file from your applications folder.
> https://apphousekitchen.com/how-to-uninstall-aldente-pro/

Have you made sure all the files have been removed ?
check the Manual Uninstallation
Make sure all the helper , support files and LaunchDaemons for AlDente are removed.
https://apphousekitchen.com/how-to-uninstall-aldente-pro/

**Have you tried reseting NPRAM as well?** 
NPRAM/PRAN also holds settings and I think **they are more related to display.**
https://support.apple.com/en-us/HT204063

Is there any other software that could conflict? 
system monitors , power management, screen controls, etc
 
Have you run Apple diagnostics?
https://support.apple.com/en-gb/HT202731

Try EtreCheck
see if you can see anything in a scan report
https://etrecheck.com/

I do know that the 2016 model can have an issue with the display ribbon cable.
Apple have acknowledged it as a fault and have a repair program.
I know this would be a coinsidence but it is still possible.  Also it mainly causes issues with the backlight
but I presume the light sensor uses that cable as well. Does not explain the battery drain though. 
https://support.apple.com/13-inch-macbook-pro-display-backlight-service

I hope this helps, well, a start at least.

RahulBhalley, 2022-03-23:
So I did all those tests but no errors were shown in any test. And have the same brightness issues. 

> I do know that the 2016 model can have an issue with the display ribbon cable.
Apple have acknowledged it as a fault and have a repair program.
I know this would be a coinsidence but it is still possible. Also it mainly causes issues with the backlight
but I presume the light sensor uses that cable as well.

It may not be a hardware issue. Because if it was then it wouldn't have needed a software (AIDente) to get triggered. 

> Does not explain the battery drain though unless it’s a separate issue.

Oh, it got fixed. Although now I need to charge battery twice / thrice a day. Possibly the battery is old and / or Apple is trying to trash Intel. Lol. 

> https://support.apple.com/en-in/13-inch-macbook-pro-display-backlight-service

I did have this issue: **Display backlight stops working completely**. My screen was dimming day-by-day and one day stopped working at all. The staff at Apple told me that screen is dead and needs to be replaced. It was back in 2019 I think. **They costed me around ₹40000 = $525**. They played me I think. Lol. Well nvm. 

> I hope this helps, well, a start at least.

I really appreciate you put your time into this. I'll possibly kill this GH issue. Will try to sell my MBP without fixing this. This issue is weirdly annoying.

closed 2022-03-25 by RahulBhalley