AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #533: Crash while Calibrating

The70sWoodsman, 2022-01-05:
**Describe the bug**
The app quite unexpectedly, generating an Apple error report whilst it running a "calibration" (it was either in the final 1hr hold or had just finished and want discharging).

**To Reproduce**
It's not possible to define how to recreate the problem as no user action was taken, it just quit unexpectedly and generated an error report.

**Expected behavior**
The app only quits when a user action to quit is executed.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
Please see the full report for most of this.
Pro Settings (true):
- Open on launch
- Sail
- Heat protection
- stop charging when powered off
- battery icon
- % icon

**Additional context**
Apple error report:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               AlDente [621]
Path:                  /Applications/AlDente.app/Contents/MacOS/AlDente
Identifier:            com.apphousekitchen.aldente-pro
Version:               1.12 (19)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-01-05 16:03:26.7932 +0000
OS Version:            macOS 12.1 (21C52)
Report Version:        12
Bridge OS Version:     6.1 (19P647)
Anonymous UUID:        1A2A8BCF-28E7-EB0A-2DBC-1AF779B44DBC

Sleep/Wake UUID:       B84BC010-E7A6-4169-8B75-F30C006D9D2E

Time Awake Since Boot: 22000 seconds
Time Since Wake:       3603 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 4 Illegal instruction: 4
Terminating Process:   exc handler [621]

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   AlDente                       	       0x1007b7252 0x100780000 + 225874
1   AlDente                       	       0x1007a4a07 0x100780000 + 150023
2   AlDente                       	       0x1007b22be 0x100780000 + 205502
3   AlDente                       	       0x1007af002 0x100780000 + 192514
4   AlDente                       	       0x100782d85 0x100780000 + 11653
5   AlDente                       	       0x1007c27e4 0x100780000 + 272356
6   Foundation                    	    0x7ff8163866c0 __NSFireTimer + 67
7   CoreFoundation                	    0x7ff8154f884f __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 20
8   CoreFoundation                	    0x7ff8154f833e __CFRunLoopDoTimer + 927
9   CoreFoundation                	    0x7ff8154f7e9d __CFRunLoopDoTimers + 307
10  CoreFoundation                	    0x7ff8154de45f __CFRunLoopRun + 1985
[... 336 lines cut ...]
}

Model: MacBookPro15,2, BootROM 1715.60.5.0.0 (iBridge: 19.16.10647.0.0,0), 4 processors, Quad-Core Intel Core i7, 2.7 GHz, 16 GB, SMC 
Graphics: Intel Iris Plus Graphics 655, Intel Iris Plus Graphics 655, Built-In
Display: 2757M, 1920 x 1080 (1080p FHD - Full High Definition), Main, MirrorOff, Online
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, LPDDR3, 2133 MHz, Micron, MT52L1G32D4PG-093
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, LPDDR3, 2133 MHz, Micron, MT52L1G32D4PG-093
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: Jul 12 2021 19:26:30 version 9.30.464.0.32.5.76 FWID 01-45ccefcd
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB2.0 Hub
USB Device: HD Pro Webcam C920
USB Device: USB Billboard Device
USB Device: USB3.0 Hub
USB Device: T2Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5

MatthiasKerbl, 2022-01-07:
Hi @The70sWoodsman,

Thank you for sharing this issue.

We are currently redoing the Calibration Mode completely and this should fix this bug. The new Calibration Mode will be released with the next update soon.

Cheers,
Matthias

MatthiasKerbl, 2022-01-16:
Hi @The70sWoodsman,

We have just released a new version of AlDente which should fix this issue.
You can download the newest version [here](https://github.com/davidwernhart/AlDente/releases) or just do the update within the app.

Cheers,
Matthias

closed 2022-01-16 by MatthiasKerbl