AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #534: Not working at all

Tuumapea, 2022-01-06:
Tried reinstalling and everything. Does not matter what you press it just keeps charging all the time. Even if I let it discharge and set the chage limit to 70% it still goes to 100 like Drake. 
Device: macOS 12.0.1 - M1 2020 Pro 13"

laurazard, 2022-01-08:
Can confirm, same thing is happening to me

djb902, 2022-01-12:
Same here... It used to work good until a few days ago. I uninstalled (manually) and reinstalled AD. No effect whatsoever. Even resetting SMC and PRAM didn't help (several times). Attached the debug file:

FYI:
MacBook Pro (Retina, 15-inch, Mid 2015)
OS: 12.1 (21C52)
Version Al Dente: 12.1

[aldentepro-debug-2022-01-11-21:59:41.pasta.zip](https://github.com/davidwernhart/AlDente/files/7852880/aldentepro-debug-2022-01-11-21.59.41.pasta.zip)

algolov, 2022-01-12:
I also confirm.
MacBook Pro M1 2020 Pro 13"
OS: 12.1 (21C52)
Version Al Dente: [12.1]
[aldentepro-debug-2022-01-12-20:36:23.pasta.zip](https://github.com/davidwernhart/AlDente/files/7856588/aldentepro-debug-2022-01-12-20.36.23.pasta.zip)

TouriM, 2022-01-13:
I also can confirm this.
I did reinstall the helper without any effect.

MacBook Air (M1 2020)
OS: 12.0.1 (21A559)
Version Al Dente: [Pro 1.12]

I'll try the tipp from djb902 later.

widewhale, 2022-01-13:
I also can confirm. It worked before but stopped at some time (1-2 weeks ago maybe). I even downgraded it to 1.11 but still no effect. 

MacBook Pro (M1 Pro 16" 2021)
OS: 12.1 (21C52)
Version Al Dente: [Pro 1.12]
[aldentepro-debug-2022-01-13-19:03:05.pasta.zip](https://github.com/davidwernhart/AlDente/files/7864091/aldentepro-debug-2022-01-13-19.03.05.pasta.zip)

stanleydesu, 2022-01-14:
Can confirm, but for when I quit the app (works if I keep the app open in the background).
M1 Air 16GB RAM 256GB SSD
OS: 12.0.1
AlDente: 1.12

GvXV3toYUNbmP, 2022-01-17:
Can anyone still reproduce this issue with newer version 1.13 ?

djb902, 2022-01-17:
Thanks for the update. I will let you know the results by the end of the day 🍀

suijkerwerk, 2022-01-19:
I can confirm also that it does not work anymore. Very annoying!

widewhale, 2022-01-19:
Updating to 1.13 didn't help. BUT. After that I restarted my Mac (1 month uptime) and it worked. So I'm not sure if it was fixed by the update. Maybe it's broke because of updating from 1.11 to 1.12 (or 1.12 to 1.13 for some people) and not restarting computer? Or it's breaking after long uptime (or because of some obscure process under the hood)?

djb902, 2022-01-19:
> I can confirm also that it does not work anymore. Very annoying!

1.13 solved the issues with me...

laurazard, 2022-01-22:
The issue is also fixed for me.

GvXV3toYUNbmP, 2022-01-22:
if not working, manually uninstall/install may work https://apphousekitchen.com/how-to-uninstall-aldente-pro/

davzaman, 2022-02-04:
Reinstalling does not work. AlDente does not work for me at all. MacOS 12.1  - MacBook Pro (15-inch, 2018)

williamhCode, 2022-02-08:
[image]
Only 1.12 works consistently for me, but not 1.13 and 1.14. The image is using 1.14, and I have left cable in for a while.

MatthiasKerbl, 2022-02-25:
> Tried reinstalling and everything. Does not matter what you press it just keeps charging all the time. Even if I let it discharge and set the charge limit to 70% it still goes to 100 like Drake.
> 
> Device: macOS 12.0.1 - M1 2020 Pro 13" App 1.12
> 
> The issue started after updating to 1.12

Hi @Tuumapea,

Do you still experience this issue with our newest version AlDente 1.14?

Cheers,
Matthias