AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #536: Helper not installing for Pro

derekhongg, 2022-01-06:
I just bought the Pro Version of Aldente and it won't install at all. Entered the Key Code and everything, but I keep getting the same pop up screen to install the helper. I've installed it over 10 times now. Says it successfully installed. Nothing still works. 

Working on 2020 Macbook Pro 13 in
OS Monterey 12.01
Aldente Pro 1.01
Disabled the optimize battery too.

MatthiasKerbl, 2022-01-07:
Hi @derekhongg,

Thank you for purchasing AlDente Pro.
Please try to install the newest version of AlDente Pro (version 1.12).
You can download the newest version here: https://github.com/davidwernhart/AlDente/releases/download/1.12/AlDente.dmg

Cheers,
Matthias

derekhongg, 2022-01-07:
Hi @MatthiasKerbl,

The link you sent me is for the AlDente Free Version. Where could I get the Pro version?

MatthiasKerbl, 2022-01-07:
> Hi @MatthiasKerbl,
> 
> The link you sent me is for the AlDente Free Version. Where could I get the Pro version?

AlDente Free and AlDente Pro have the same program file. However, you have to activate AlDente with your license key in order to be able to use the Pro features.

derekhongg, 2022-01-07:
@MatthiasKerbl ahhh, that fixed it. Thank you!

closed 2022-01-07 by MatthiasKerbl