AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #541: Support "Shortcuts"

notDavid, 2022-01-13:
Hi there,

please consider implementing support for the [Shortcuts](https://support.apple.com/guide/shortcuts/welcome/ios) app, to set the `charge limit`, and possible other settings.
This would make automation possible in many ways, and also solve some of the other feature requests in this repo.

Thank you!

MatthiasKerbl, 2022-01-13:
Hi @notDavid,

You are right, implementing Shortcuts would be a great addition to AlDente. We have already started looking into Shortcuts but since it is a bigger development effort, it will take time.

Cheers,
Matthias

Fir333, 2022-08-16:
Would be great if we could minimize al dente to tray :) Got enough of stuffs when i press Command + tab ;) Would be great to miss al dente here <3

MatthiasKerbl, 2022-08-17:
> Would be great if we could minimize al dente to tray :) Got enough of stuffs when i press Command + tab ;) Would be great to miss al dente here <3

Hi @Fir333,

You just have to deactivate the dock icon in Settings/Appearance/Show Dock Icon.

Cheers,
Matthias

MatthiasKerbl, 2022-08-17:
Hi @notDavid, did not know that. Thanks for mentioning it.

MatthiasKerbl, 2022-08-20:
Hi @notDavid,

We released a new version of AlDente Pro a couple of days ago which includes support for Apple Shortcuts. Please keep in mind that this is still a beta feature and bugs may occur. Also, check out the article about this new feature: https://apphousekitchen.com/feature-explanation-shortcuts/

Cheers,
Matthias

closed 2022-09-01 by MatthiasKerbl