AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #543: background mode

jjangga0214, 2022-01-14:
Hi. Thanks for maintaining this project.

When I `cmd` + `tab`, AlDente appears up, as it's running in foreground mode.
I want AlDente to be like a demon, which only appears on the top status bar.

Sincerely.

joujek, 2022-01-14:
As far i know app asks for  install deamon as the very first step afer first launch.
Im other hand the app Windows triggers icon on topbar as well
Do you install latest version correctly?

MatthiasKerbl, 2022-01-15:
> Hi. Thanks for maintaining this project.
> 
> When I `cmd` + `tab`, AlDente appears up, as it's running in foreground mode. I want AlDente to be like a demon, which only appears on the top status bar.
> 
> Sincerely.

Hi @jjangga0214,

You have to activate the dock icon in order for AlDente to be in the background completely.
Go to Settings/Appearance/Other and disable "Show Dock Icon".

Cheers,
Matthias

closed 2022-01-15 by MatthiasKerbl

adamigen, 2025-04-08:
Hi @MatthiasKerbl ,
I have activated the dock icon as you suggested, and AlDente still appears in the cmd+tab menu. If I close it, the entire application exits, including the dock icon.
How can I have the application running only in dock icon mode, without the clutter in cmd+tab ?

Thanks

yeison, 2025-05-11:
> Hi [@jjangga0214](https://github.com/jjangga0214),
>
> You have to activate the dock icon in order for AlDente to be in the background completely. Go to Settings/Appearance/Other and disable "Show Dock [...]

I see.  Thanks for letting us know how to put it in background mode.  I think this setting is a little bit confusingly worded.  Is there a more clear way to word it to make it more user friendly?  Thanks for the great app.