AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #544: Al Dente Pro claims it can't connect to the Internet and won't activate my license

rkeniger, 2022-01-15:
**Describe the bug**
I've been using AlDente Pro for a while and it's been fine. I just rebooted my Mac and now it says:

"AlDente needs to check if your license is valid once every 10 days. Please connect to the internet for AlDente to work."

The trouble is, my internet connection is working just fine. I've restarted the app multiple times and clicked "Activate" and keep getting the same message.

I'm not running a VPN or proxy or anything else that might limit the app.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch AlDente
2. See message

**Expected behavior**
My license should be activated

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model: MacBook Pro (14-inch, 2021) 
 - macOS version: Monterey 12.1 (21C52)
 - AlDente Pro
 - AlDente Version: 1.12
 - List of all enabled/disabled settings in AlDente: N/A

rkeniger, 2022-01-15:
I just used Charles to check out what's happening and it looks like v3.paddleapi.com is returning a 503

MatthiasKerbl, 2022-01-15:
Hi @rkeniger,

Thanks for reporting.
We are already aware of this bug and currently in contact with our licensing partner to get it fixed.

For now, a quick workaround should be as follows:

1. Go into Settings/General Settings
2. Click on "Manage License"
3. A popup window should open, click on "Deactivate License"
4. Click on "Activate License" and enter your email address and license code again

I will post follow-ups regarding this bug on the first post about this issue. #500 

If you notice anything else, let us know.

Cheers,
Matthias

closed 2022-01-15 by MatthiasKerbl

rkeniger, 2022-01-15:
Thanks @MatthiasKerbl, that worked perfectly.

avxkim, 2022-02-20:
@MatthiasKerbl any news on this? Just faced this issue after restart...

notDavid, 2022-03-31:
[image]

 @MatthiasKerbl Please reopen this issue, it's still happening... sigh.

MatthiasKerbl, 2022-03-31:
> 

Hi @notDavid,

I am sorry that it happened again to you.
We did everything we could to lower the chances of it happening. Unfortunately, we have to wait for our licensing partner to fix their bugs for it to never happen again.

Cheers,
Matthias

notDavid, 2022-03-31:
@MatthiasKerbl I doubt Paddle will ever "fix" this, i assume it's a feature: tracking customers. They want to gather analytics... and this way they force it.

MatthiasKerbl, 2022-04-01:
> @MatthiasKerbl I doubt Paddle will ever "fix" this, i assume it's a feature: tracking customers. They want to gather analytics... and this way they force it.

Hi @notDavid,

When we brought this issue to Paddle, they told us they found a bug in their system and they will fix it. Unfortunately, they could not give us a timeline.