AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #545: *HELP* I have miss entered my email address while purchasing the pro version

HRamadan1, 2022-01-15:
I have written my email address with one letter misspelled and now I can't reach my (already paid subscription ) and can't reach support because support is only for pro members !!!
what should I do :(

york601, 2022-01-16:
I left my email and paid but didn't receive the letter what even clicked "restore your authorization code" @MatthiasKerbl

MatthiasKerbl, 2022-01-16:
> I have written my email address with one letter misspelled and now I can't reach my (already paid subscription ) and can't reach support because support is only for pro members !!! what should I do :( @MatthiasKerbl

Hi @HRamadan1,

No worries, just send me the email you used to purchase AlDente Pro to news@apphousekitchen.com

Cheers,
Matthias

HRamadan1, 2022-01-17:
@MatthiasKerbl  Thank you so much 
may you kindly check the email 🙏

york601, 2022-01-17:
***@***.***.That was the email, while I used to purchase AlDente Pro.
  Looking forward to your reply.

发自我的iPhone

MatthiasKerbl, 2022-01-17:
Hi @HRamadan1 and @york601,

I have forwarded the payment emails to the correct email addresses now.

Cheers,
Matthias

closed 2022-01-17 by MatthiasKerbl