AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #547: [Feature request] Smaller menubar icon

nheeren, 2022-01-16:
With the introduction of the notch in the new Macbook Pros, menubar estate has become very scarce. Would it be possible you offered a narrower menubar icon?

closed 2022-01-17 by MatthiasKerbl

nheeren, 2022-01-17:
Thanks for the feedback! Too bad. I wouldn't want to lose the icon completely. 

In my humble opinion not only the icon is quite large, it also seems to have quite a spacer left and right.  Maybe turn the battery 90 degrees? ;)
[image]

huyz, 2022-01-17:
I did also notice that there is an unusual amount of whitespace to the left and right of the Al Dente icon, unlike the icons of other apps.

MatthiasKerbl, 2022-01-17:
> Thanks for the feedback! Too bad. I wouldn't want to lose the icon completely.
> 
> In my humble opinion not only the icon is quite large, it also seems to have quite a spacer left and right. Maybe turn the battery 90 degrees? ;) [image]

Now I know what you mean. We will look into this, maybe we can reduce the space left and right of the icon.

nheeren, 2022-01-30:
Maybe you can also consider to display the battery charge / percentage (e.g. as a color bar in the background). This would allow me to ditch the macOS / iStat battery icon instead and also save space ;)

huyz, 2022-01-30:
@nheeren Already possible:
[image]

nheeren, 2022-01-30:
Thanks, but activating the battery icon will take even twice the space. I am trying to suggest a more multi-functional single icon, in case the developers don't want to offer a smaller icon.