AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #550: Can you redesign the UI of AIDente Pro?

fhefh2015, 2022-01-17:
Can you redesign the UI of AIDente Pro?

It feels like it takes up too much space.

Can you be a little more concise?

[image]

MatthiasKerbl, 2022-01-17:
> 

Hi @fhefh2015,

Do you mean the menubar icon?
You can completely customise the appearance of the menubar icon in Setting/Appearance.

Cheers,
Matthias

fhefh2015, 2022-01-17:
@MatthiasKerbl 

This pop-up window(view)
[image]

Can be designed like this, simple and more

https://github.com/MonitorControl/MonitorControl

[image]

MatthiasKerbl, 2022-01-17:
> @MatthiasKerbl
> 
> This pop-up window(view) [image]
> 
> Can be designed like this, simple and more
> 
> https://github.com/MonitorControl/MonitorControl
> 
> [image]

Hi @fhefh2015,

We are fairly satisfied with the current design. Therefore, I don’t think we will update it soon and concentrate more on improving and adding features. However, thank you for the inputs and we are of course open to doing something in this direction in the future.

Cheers,
Matthias

closed 2022-01-17 by MatthiasKerbl