AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #552: Can we get the option to view a "real charge level"

PearApple151, 2022-01-17:
I use two apps to manage my battery now. one that tracks my usage, cycles, capacity, etc. and Al Dente. the only issue is that Al Dente shows the battery life that my Mac believes it has. the third party app shows me battery life as a percent of current charge / current capacity. I don't know how Mac calculates my battery life, but I've noticed from the beginning it was very optimistic. sit at 100% for like an hour of use. meanwhile any third party app will say I'm at 94ish% after an hour. Right now, Al Dente says I'm at 98% and my third party app says I'm at 92.6%.  I trust the third party apps more for my percent, but my third party app isn't responsible for stopping my charging. Could Al Dente calculate percent like that?

MatthiasKerbl, 2022-01-18:
> I use two apps to manage my battery now. one that tracks my usage, cycles, capacity, etc. and Al Dente. the only issue is that Al Dente shows the battery life that my Mac believes it has. the third [...]

Hi @PearApple151,

You are right, macOS uses a different battery percentage than the "real" one which the battery reports. A feature for switching between macOS and "real" battery percentage is already on our list for future updates and we will start implementing it in one of the upcoming updates.

Cheers,
Matthias

MatthiasKerbl, 2022-04-12:
Hi There,

we have just released AlDente 1.16 which includes a new feature called "Hardware Battery Percentage". If enabled, AlDente will use the battery percentage which the battery management system reports.

You can do the update in AlDente or download the newest version here: https://github.com/davidwernhart/AlDente/releases

Cheers,
Matthias

closed 2022-04-12 by MatthiasKerbl