AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #553: Aldente does not work at all on my Mac

dogyoonlee, 2022-01-19:
**Describe the bug**
Charge limit does not work at all on both ClamShell mode and normal mode. 
I set the limit as 75%, but it does not work at all. 
The battery have been charged until 100% regardless of AlDente setting.

**To Reproduce**
Steps to reproduce the behavior:

1. Just install the AlDente.
2. Does not work.

**Expected behavior**
Charge limit works. 

**Screenshots**
[image]
[image]
[image]

**MacBook (please complete the following information):**
 - MacBook Pro 14' 2021, M1 Pro.
 - Momentary 12.1
 - AlDente Free.
 - 1.13 (But 1.12 was not work as same as latest version)
 - charge limit disable.

**Additional context**

williamhCode, 2022-01-19:
It seems like Aldente stopped working on my Macbook Pro 13' M1 after I updated Aldente to 1.13. I removed and installed 1.12 and it works.

jeanqui, 2022-01-19:
Same here. Though 1.12 was working for me. (I'm on the pro plan but I dont think that should make a difference)

dogyoonlee, 2022-01-20:
@williamhCode @jeanqui Thanks!! it works for me too. 
I close this issue!

closed 2022-01-20 by dogyoonlee

jeanqui, 2022-01-20:
Wait should the issue be closed? because it is still an issue with the new update?