AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #554: [Feature Request] Why not provide a illustrative graph for the charging process

hsir, 2022-01-19:
Just a feature idea (maybe for the Pro version).... Because a lot of people may get confused about the app functionalities when they hand on this app for the first time, so maybe the app can provide a detailed graph (just like the native battery graph view in the macOS preferences panel) of the past 24 hours, indicating and showing the charing process, and providing some brief explanations of each stages (e.g. the sailing mode stage, the charging stage, and the calibration stage). This will make users get more confident about keeping their batteries charged under this utility app.

Hope this can be helpful.

MatthiasKerbl, 2022-01-19:
Hi @hsir,

Thank you for your input. This sounds really helpful. However, I am afraid it is rather difficult to implement because every time your MacBook is in sleep or powered off, third-party apps can’t collect any data.
One option would be to implement graphs like that in the feature explanations on our blog to make the charging behavior more understandable. https://apphousekitchen.com/category/featureexplanation/

Qhilm, 2024-06-20:
My use case for this feature: I use lots of different USB chargers and sometimes, some are not powerful enough and the MacBook is draining instead of charging because there is not enough power delivered over USB. But it takes me a while to notice because I do not memorise the percentage level and hence I don't see what the trend is.

The "live" power flow is very useful, but since the power consumption on the mac may vary a lot, it's only a moment in time, laptop could be charging fine one moment and then depleting the next. A small historical graph would provide a clear visual trend.

And to comment on the "sleep issue", I think it still would have a lot of value even if it would display gaps when the MacBook was sleeping. At least I would see a graphical representation of the charging and when the laptop is sleeping I'm really not worried about the direction of the charging, I know it will charge and not deplete.

And maybe a "full in X hours" (based on stats since last time the laptop went to sleep or last time the laptop was connected to power or last 10 minutes, whatever is the shortest) would help too.

A happy Pro user =)

MatthiasKerbl, 2025-02-06:
Hi @hsir,

Our newest version of AlDente has a new Dashboard feature which includes graphs about the charging process.

Best,
Matthias

closed 2025-02-06 by MatthiasKerbl