AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #555: Change in charge limit ignored until I do something dramatic

DamonHD, 2022-01-19:
**Describe the bug**
If for example I raise the charge limit in 1.12 or 1.13 above where it was charged to, charging does NOT start.  Sometimes reinstalling the helper or disconnecting and reconnecting mains power seems to help.

**Expected behavior**
When AlDenta has correctly paused charging, and then I use the UI to raise the charge limit (well) above the current charge level, I expect charging to start again.

**MacBook (please complete the following information):**
 - MacBook model MacBookAir6,2
 - macOS version 11.6.2
 - AlDente Pro or AlDente Free? Pro 
 - AlDente Version? 1.12 and 1.13
 - List of all enabled/disabled settings in AlDente: Intel Mode

**Additional context**

MatthiasKerbl, 2022-02-02:
Hi @DamonHD,

I am currently not able to reproduce this bug with my Intel based MacBooks. However, we have made some improvements to the Intel Mode with AlDente 1.14. Please update to AlDente 1.14 and let us know, if you still experience this issue.

Cheers,
Matthias

DamonHD, 2022-02-02:
I have upgraded.

Already behaviour seems better.

The slider bar is showing green or blue rather than grey, and switches between green and blue (and + pause or) within a few seconds of me moving the slider.

Also, other tools agree that the battery is charging or not as I'd expect having moved the slider.

I'll keep an eye on it!

Thank you.

Rgds

Damon

MatthiasKerbl, 2022-02-02:
> I have upgraded.
>
> Already behaviour seems better.
>
> The slider bar is showing green or blue rather than grey, and switches between green and blue (and + pause or) within a few seconds of me moving [...]

I am glad to hear that. :)

DamonHD, 2022-02-02:
Ah, no, it's gone unresponsive again.

The bar has gone grey, and pushing it up beyond the actual charge level is not letting charging restart.

I've tried reinstalling the helped and closing and restarting Al Dente.

Rgds

Damon

DamonHD, 2022-02-02:
A forced remove of the helper, and another shut down and restart of AlDente has the bar now green/+.

Rgds

Damon

MatthiasKerbl, 2022-02-02:
That’s unfortunate, could you send us a debug file when it happens the next time? https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/

DamonHD, 2022-02-02:
While it is failing to respond usefully to changes in charge limit, "Save Debug File"?

Rgds

Damon

MatthiasKerbl, 2022-02-02:
> While it is failing to respond usefully to changes in charge limit, "Save Debug File"?
> 
> Rgds
> 
> Damon

Yes, exactly. This would be really helpful.

DamonHD, 2022-02-02:
OK, it's happened again.  I've sent you the debug file.

It might be better for these files to be in a human-readable format as far as possible like for the Apple debug dumps, so we can be sure that you're not exfiltrating all our secrets in there!

Rgds

Damon

MatthiasKerbl, 2022-02-03:
Hi @DamonHD,

We have looked into your debug file and it seems that your MacBooks sometimes disconnect for a short period. One cause of this problem could be a faulty charging cable/brick. Do you have a second charger you can test it with?

DamonHD, 2022-02-03:
I doubt it's faulty.  I do disconnect it from time to time eg when I'm going to be away from the machine for a while, or to switch over to my off-grid adaptor (I may switch back and forth several times on a day with intermittent sunshine).

And I have been disconnecting and reconnecting additionally to try and provoke AlDente into paying attention to my changed charge limit.

Maybe I can leave the mains adaptor permanently connected and on (other than maybe overnight) and somehow get AlDente into a good state.

I assume that the slider bar is not meant to be grey all the time?

Rgds

Damon

MatthiasKerbl, 2022-02-05:
Hi @DamonHD,

Thanks for the detailed description.

> Maybe I can leave the mains adaptor permanently connected and on (other than maybe overnight) and somehow get AlDente into a good state, and we can run a test of some sort?

You could try to do that for a while.

> I assume that the slider bar is not meant to be grey all the time?

No, not at all, only when the charger is disconnected.

This it what the slider should look like when charging:
[image]

This it what the slider should look like when charging is paused:
[image]

This it what the slider should look like when disconnected:
[image]

Unfortunately, I am still not able at all to reproduce this bug on any Intel MacBook I have close to me.

MatthiasKerbl, 2022-04-19:
Hi @DamonHD,

Do you still encounter this issue with our newest version AlDente 1.16.1?

Cheers,
Matthias

DamonHD, 2022-04-19:
Yes.

And I'm still getting the ~30s delay to being able to log in after opening up my Mac, which I suspect is when your licence code is trying to check my licence validity but my WiFi has not reconnected.

Rgds

Damon

DamonHD, 2022-04-19:
I'm trying to re-install the helper, since I may not have done so since installing this latest point release.

As ever, ADP is hanging and will require a Force Quit for me to be able to complete the task!  %-P

Rgds

Damon

MatthiasKerbl, 2022-04-19:
Hi @DamonHD,

You should try to uninstall AlDente completely manually before installing it again:
https://apphousekitchen.com/how-to-uninstall-aldente-pro/

DamonHD, 2022-04-19:
OK, have uninstalled and reinstalled.

Rgds

Damon

DamonHD, 2022-04-19:
Oh, and the help seems to have died again (the bar has gone grey).

I have been on the mains adaptor unbroken since posting the last comment.

The screen has been blanked to save power while I was way from my desk.

Rgds

Damon

MatthiasKerbl, 2022-04-19:
Please send us a debug file immediately when the issue occurs.

DamonHD, 2022-04-19:
Have just now made and sent one.

But it happens all the time, usually very soon after reinstalling, so I'd need a hotline permanently open to keep sending!  B^>

Rgds

Damon