AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #556: Discharging while in sleep mode

VIS89, 2022-01-21:
Hello,
I have a strange issue and it started with AlDente installation and configuration.
I work on laptop 8-10 hours a day, and it was on 100% battery before AlDente. I put a laptop in sleep mode after that. On the next day when I start it, it's 100% or 98%, something like that.
When I configured AlDente to keep 60% to 70% and put the laptop in sleep mode, on the next day my battery is like 5-15%.
And I can feel the body of the laptop not so cool like before, looks like something is running in the background. When I disable AlDente everything is okay.

Mac: MacBook Pro (13-inch, 2018, Four Thunderbolt 3 Ports
OS: MacOS Monterey 12.1

JoostK, 2022-01-21:
I can confirm this behaviour on a MBP M1 Max:

[image]

The drop during the evening was me working on the machine for 4 hours, but after closing the lid and coming back to it this afternoon it had almost completely gone through its remaining charge overnight.

I am suspecting AlDente because of this in `pmset -g assetions`:

```
   pid 708(com.apphousekitchen.aldente-pro): [0x0001527f00079bf9] 00:02:30 PreventSystemSleep named: "AlDente-Pro"  
```

but that may be a red-herring, of course.

These are my settings:

[image]

MatthiasKerbl, 2022-01-22:
Hi @JoostK and @VIS89,

Thank you for your issue report. We are aware that some users experience this issue and we are currently investigating it. Hopefully, we can fix it soon with an update.

Proplex, 2022-01-23:
Just chiming in to say I'm also experiencing the same issue after updating to 1.13

Zer0xPoint, 2022-01-26:
same here
12.1 monterey
2021 16'' m1 pro
10% drop over night

razvan-negri, 2022-01-27:
Same behavior, sent debug logs to David via email.

layandreas, 2022-01-27:
I have the same issue. For me the problem is the screen stays on when closing the lid. I think it may be related to this issue: https://github.com/davidwernhart/AlDente/issues/560

lbdsgn, 2022-02-01:
I will occasionally wake up to a 0% dead battery after it's been asleep all night. It just happened. I suspect Al Dente. MacBook Pro (16-inch, 2019)

https://share.getcloudapp.com/4guZ4Arb

(I will often also find my battery at 100% after sleeping plugged in, though it's set for 80% max)

xawill, 2022-02-01:
Could it be a macOS bug ?
https://www.macrumors.com/2022/01/30/macos-12-2-bluetooth-battery-drain/

MatthiasKerbl, 2022-02-01:
> I will occasionally wake up to a 0% dead battery after it's been asleep all night. It just happened. I suspect Al Dente. MacBook Pro (16-inch, 2019)
> 
> https://share.getcloudapp.com/4guZ4Arb
> 
> (I will often also find my battery at 100% after sleeping plugged in, though it's set for 80% max)

Hi @lbdsgn,

Regarding the battery drain, this might be due to a bug of AlDente and we will release an update soon.
When it comes to your battery being at 100% after sleeping: Depending on you using AlDente Free or AlDente Pro and which features you have enabled, AlDente handles this situation differently. Check out our blog post about that here: https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/

Cheers,
Matthias

MatthiasKerbl, 2022-02-01:
> Could it be a macOS bug ? https://www.macrumors.com/2022/01/30/macos-12-2-bluetooth-battery-drain/

Interesting, thanks for sharing.

lbdsgn, 2022-02-02:
I am on 12.0.1, so probably this bluetooth bug for 12.2 is not the cause?

lbdsgn, 2022-02-02:
> 

Thanks Matthias I am using the pro/paid version.

MatthiasKerbl, 2022-02-02:
Hi everyone,

We have just released a new version of AlDente (1.14) which should fix the issues you experienced.
If you still encounter any of the issues after updating, please report back.

Download AlDente 1.14 here (https://github.com/davidwernhart/AlDente/releases) or update within the app.

Thank you for your patience.

Cheers,
Matthias

closed 2022-02-02 by MatthiasKerbl

whateverthatis, 2024-10-06:
I'm just having the same issue on version 1.28.4:

[image]

 I woke up to a dead battery after a 70% charge on a brand new MacBook Air M3 13", macOS Sequoia.

MatthiasKerbl, 2024-10-07:
> I'm just having the same issue on version 1.28.4:
> 
> <img alt="Screenshot 2024-10-06 at 11 39 13" width="469" src="https://private-user-images.githubusercontent.com/108505953/373936743-1e401026-4d1a-4dee-8b29-371314db684a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgyODYwODUsIm5iZiI6MTcyODI4NTc4NSwicGF0aCI6Ii8xMDg1MDU5NTMvMzczOTM2NzQzLTFlNDAxMDI2LTRkMWEtNGRlZS04YjI5LTM3MTMxNGRiNjg0YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwN1QwNzIzMDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lNWExMzU5ODMxOWQ4MmIzNDRhYjBkNjA4NmQ4Mjc2YjVkMjA4MjYyZjUwNTBmNDMwN2E5OTU4MjMyZWQwOTM5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.ksBcVXIb9LKLcVkWd3s__fl9pHoZc8XuBfxJoEYDApg">
> I woke up to a dead battery after a 70% charge on a brand new MacBook Air M3 13", macOS Sequoia.

Hi @whateverthatis,

If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.

Best,
Matthias

MatthiasKerbl, 2024-10-07:
> > > I'm just having the same issue on version 1.28.4:
> > > <img alt="Screenshot 2024-10-06 at 11 39 13" width="469" src="https://private-user-images.githubusercontent.com/108505953/373936743-1e401026-4d1a-4dee-8b29-371314db684a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgyODYwODUsIm5iZiI6MTcyODI4NTc4NSwicGF0aCI6Ii8xMDg1MDU5NTMvMzczOTM2NzQzLTFlNDAxMDI2LTRkMWEtNGRlZS04YjI5LTM3MTMxNGRiNjg0YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwN1QwNzIzMDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lNWExMzU5ODMxOWQ4MmIzNDRhYjBkNjA4NmQ4Mjc2YjVkMjA4MjYyZjUwNTBmNDMwN2E5OTU4MjMyZWQwOTM5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.ksBcVXIb9LKLcVkWd3s__fl9pHoZc8XuBfxJoEYDApg">
> > > I woke up to a dead battery after a 70% charge on a brand new MacBook Air M3 13", macOS Sequoia.
> > 
> > 
> > Hi @whateverthatis,
> > If possible, please share a debug file the next time this happens so that we can assess this issue better. You can find a guide on [How to generate and share a debug file](https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/) on our blog.
> > Best, Matthias
> 
> Hi Matthias,
> 
> I'm unable to share the debug file in GiHub, as it does not allow me to attach .pasta files... Is there any other way?
> 
> The bug happens all the time, if I unplug my mac the battery begins to drain, unless I quit the AlDente app.
> 
> Thank you for looking into this.
> 
> Best,
> 
> Miklós

Hi @whateverthatis,

> Is there any other way?

Yes, you can just add ".txt". Afterward, you should be able to upload it here on GitHub.

> The bug happens all the time, if I unplug my mac the battery begins to drain, unless I quit the AlDente app.

This is odd because we have multiple mechanisms in place to avoid such issues from happening. We will be able to know more with the debug file.

Best,
Matthias

whateverthatis, 2024-10-07:
> Hi @whateverthatis,
>
>
> Yes, you can just add ".txt". Afterward, you should be able to upload it here on GitHub.
>
>
> This is odd because we have multiple mechanisms in place to avoid such issues from [...]

Hi Matthias,

I've attached the debug file. Thank you.
[aldentepro-debug-2024-10-06-11:31:13.pasta.txt](https://github.com/user-attachments/files/17283096/aldentepro-debug-2024-10-06-11.31.13.pasta.txt)

Best,

Miklós,

whateverthatis, 2024-10-08:
Hi Matthias,

Further weird things are happening with my battery:
[image]

Here are my settings:
[image]
[image]

And here is the latest debug file:
[aldentepro-debug-2024-10-08-09:26:04.pasta.txt](https://github.com/user-attachments/files/17289568/aldentepro-debug-2024-10-08-09.26.04.pasta.txt)

Cheers,

Miklós

Razer0123, 2025-01-14:
Any update on this? I think I have a similar issue and may be caused by two options, auto discharging and disable charging while asleep

Any chance they are keeping the Mac awake during sleep?

arthurgeron, 2025-06-11:
I have aldente installed on two machines, this annoying bug happened on both.. I've uninstalled it from one and now the bug no longer happens on that machine. 
It's very unflattering that this issue isn't being taken seriously, given it's a paid product. I'll soon ask for a refund, as it clearly isn't going to be fixed anytime soon.

MatthiasKerbl, 2025-06-12:
Hi @Razer0123 and @arthurgeron,

I'm sorry to hear you're having issues with AlDente. In general, we are always trying to make AlDente as stable as possible. From our experience, issues like that are most often not caused by AlDente, and we usually recommend running [Sleep Aid](https://ohanaware.com/sleepaid/), which offers a free trial, for a couple of days to determine the cause of the issue. If the self-discharge is caused by AlDente, we are eager to fix it as soon as possible.

Best,
Matthias