AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #559: Switching user accounts does not respect then charge limit

rka1980, 2022-01-21:
**Describe the bug**
Using AlDente free on a MacBook Air M1 with 2 user accounts. Both user accounts run Ardent with a limit of 89%. When I'm in account A the charge limit is respected. When I switch to account B while account A is still signed in, the charge limit is not respected and the battery will be charged to 100%. When I switch back to account A it detects that the charge is already over the limit and stops charging. The problem here is that when using account B the battery will be charged beyond the charge limit. Every time when this happens I have to activate the discharge option every time when switching accounts. This makes the application kind of useless.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Expected behaviour is that the charge limit will be respected when switching to another account on the MacBook when AlDente is active on that other account as well.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model MacBook Air M1 2020
 - macOS version macOS Monterey 12.1
 - AlDente Pro or AlDente Free? AlDente Free
 - AlDente Version? 1.13
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Add any other context about the problem here.

rka1980, 2022-01-25:
Reverting back to v1.12 solved the issue.

samwzlim, 2022-02-22:
Facing this issue as well on the latest version (v1.14). Is there a solution to this issue other than reverting back to v1.12? I wish to be on the latest version of AlDente at all times.

**MacBook (please complete the following information):**

- MacBook model: MacBook Pro M1 Pro
- macOS version: macOS Monterey 12.2.1
- AlDente Pro or AlDente Free? AlDente Free
- AlDente Version? 1.14
- List of all enabled/disabled settings in AlDente: Launch at login selected

MatthiasKerbl, 2022-03-01:
Hi everyone,

Thank you for reporting this issue.
We are currently working on an update that will add the fast user switching functionality.
It will be released as soon as possible.

Thank you for your patience.

Cheers,
Matthias

MatthiasKerbl, 2022-03-03:
Hi everyone,

We have just released AlDente 1.15 which finally supports Fast User Switching.
You can just do the update in AlDente or download the newest version here: https://github.com/davidwernhart/AlDente/releases

Also, please read the tutorial and explanation of how to get the Fast User Switching to work properly here: https://apphousekitchen.com/fast-user-switching-explanation-and-tutorial/

Hope you like it.

Cheers,
Matthias

closed 2022-03-03 by MatthiasKerbl