AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #560: Aldente disable sleep option in main menu

sparvaneh, 2022-01-21:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'apple logo' on top left
2. Scroll down to 'Sleep'
4. 'Sleep' is grayed out and can't press it to put laptop in the sleep mode

**Expected behavior**
The sleep option was active in the previous version

**MacBook (please complete the following information):**
 - MacBook model (MacBook Pro (15-inch, 2016))
 - macOS version (Monterey)
 - AlDente Pro or AlDente Free? AlDente Free
 - AlDente Version?  1.13

cheahm, 2022-01-22:
Same issue here - led to unexpected heat/drain when I had packed up my laptop the other day!

Reverted to 1.12 for now and sleep option immediately is restored.

MacBook Pro 14 (M1 Pro)
macOS Monterey 12.1
AlDente Free 1.13

MatthiasKerbl, 2022-01-22:
Hi @sparvaneh,

In which situation is the sleep option grayed out?
While your MacBook is connected to the power adapter?
While it is charging?
While it is connected with the power adapter and charging is paused by AlDente?
While it is disconnected?
While discharging with AlDente?

linroex, 2022-01-22:
same here

Macbook Pro 14" (M1 Pro)
macOS 12.1
AIDente Pro 1.13
when it is connected power adapter but charging

sparvaneh, 2022-01-23:
Sleep mode is grayed out when connected to the power adaptor and when on battery.

GvXV3toYUNbmP, 2022-01-23:
I thought this issue was by design from previous version and fixed on latest version https://github.com/davidwernhart/AlDente/issues/495#issuecomment-1001965693

junian, 2022-01-24:
Same with me. You can check that AlDente is preventing sleep from terminal:

```bash
junian@MacBook-Air-M1-2020 ~ % pmset -g
System-wide power settings:
 SleepDisabled		0
Currently in use:
 standby              1
 Sleep On Power Button 1
 hibernatefile        /var/vm/sleepimage
 powernap             1
 networkoversleep     0
 disksleep            10
 sleep                1 (sleep prevented by coreaudiod, coreaudiod, com.apphousekitchen.aldente-pro, powerd)
 hibernatemode        3
 ttyskeepawake        1
 displaysleep         10 (display sleep prevented by com.apphousekitchen.aldente-pro)
 tcpkeepalive         1
 lowpowermode         0
 womp                 1
```

Once I quit AlDente, it's gone from sleep preventer.

VivekHalakatti, 2022-01-24:
hello, i have been facing the same issue with Aldente.It greys out the sleep button rendering your m1 macbook unable to sleep. I have a fix for that!, try this until Aldente folks release a bug fix update.
<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01//EN" "http://www.w3.org/TR/html4/strict.dtd">
<html>
<head>

<meta http-equiv="Content-Style-Type" content="text/css">
<title></title>
<meta name="Generator" content="Cocoa HTML Writer">
<meta name="CocoaVersion" content="2113.2">
<style type="text/css">
p.p1 {margin: 0.0px 0.0px 0.0px 0.0px; font: 13.0px 'Helvetica Neue'}
p.p2 {margin: 0.0px 0.0px 0.0px 0.0px; font: 13.0px 'Helvetica Neue'; min-height: 15.0px}
table.t1 {border-collapse: collapse}
td.td1 {border-style: solid; border-width: 1.0px 1.0px 1.0px 1.0px; border-color: #9a9a9a #9a9a9a #9a9a9a #9a9a9a; padding: 1.0px 5.0px 1.0px 5.0px}
</style>
</head>
<body>
<p class="p1">Run this in terminal:-</p>
<p class="p1">1. Log in with an account that has admin privileges</p>
<p class="p2"><br></p>
<p class="p1">2. Open Terminal and run the following command:</p>
<p class="p2"><br></p>

1 | sudo defaults write /Library/Preferences/SystemConfiguration/com.apple.PowerManagement SystemPowerSettings -dict SleepDisabled -bool NO
-- | --

<p class="p1">3. Restart the Mac (this is necessary to apply the change.)</p>
<p class="p2"><br></p>
<p class="p1">After the restart, you should see that the&#xA0;<b>Sleep</b>&#xA0;command is available again in the Apple menu.</p>
</body>
</html>

NickStrupat, 2022-01-24:
I have found that uninstalling and reinstalling the the `Helper Application` also works, but I'm not yet sure for how long.

cisasteelersfan, 2022-01-26:
I was able to enable sleep with `$ sudo pmset -a disablesleep 0`. But it soon became grayed out again.

goncalossilva, 2022-01-27:
I've hit this a couple of times, too. It happened just now after I hit "Top Up".

gitti22, 2022-01-27:
Hello,  I get the same problem:
I use Al Dente Pro 1.13 on Macbook Air M1 with Mac OS 12.2.
I realized following problem:
If the battery is more than the charge limit (my setting 80%) and I put the button "Discharge" then the macbooks sleep mode is turned of (that is: it is not possible to turn it on via "apple"-icon in the top left corner of the screen. there sleep mode (German: "Ruhezustand") is just shown in grey (not black, as all other command lines).
This is not changing anymore, if I switch of "Discharge".
Even, if I shut down the Macbook it is not changing. if I sign in as the (usual) standard user (or as an admin user; or in secured mode). 
(Often also the error report (translation from German:) "Last time you quit the computer due to existing problems. Should the previously opened programs be reopened?" shows up.)

One time I could solve the problem (I do not know why and could not replicate it):  After shutting down and restart+log in as an admin sleep mode was active again (also after log in as standard user).

How can I stop this error?

easp, 2022-01-28:
Same Issue, Apple Menu > Sleep is greyed out 
Intel MacBook Pro running MacOS 12.2 
AlDente Free 1.13
Observed when Powered, not charging, at my configured charging limit. Also observed when on battery.
Aldente not listed as blocking sleep in pmset -g.
Problem persisted across a reboot.

gameinthebox, 2022-01-28:
Chiming in to report the same issue here, Apple Menu > Sleep is greyed out
MacBook Pro 14" M1 Pro
macOS 12.2 (21D49)
AlDente Free 1.13
Observed when powered over USB-C (not MagSafe), not charging at limit while in clamshell mode.

MatthiasKerbl, 2022-02-02:
Hi everyone,

We have just released a new version of AlDente (1.14) which should fix the issues you experienced.
If you still encounter any of the issues after updating, please report back.

Download AlDente 1.14 here (https://github.com/davidwernhart/AlDente/releases) or update within the app.

Thank you for your patience.

Cheers,
Matthias

closed 2022-02-02 by MatthiasKerbl

GJRobert, 2022-02-26:
Dear developer,
I have updated to 1.14, but my `Sleep` is still grayed out. I have found only recently that my laptop has not been entering into sleep mode for perhaps one month. I'm not sure how is everyone else going with 1.14.

MatthiasKerbl, 2022-03-01:
> Dear developer, I have updated to 1.14, but my `Sleep` is still grayed out. I have found only recently that my laptop has not been entering into sleep mode for perhaps one month. I'm not sure how is everyone else going with 1.14.

Hi @GJRobert,

I answered you at this post #601

Cheers,
Matthias

itzTimmyy, 2022-10-23:
Not really sure if this is fixed or its a normal thing but my sleep button is greyed out.
The version is 1.19 and OS 12.6 MacBook Air (M2, 2022)

The sleep button is greyed out whether or not the device is plugged in or not to power.

I used terminal and inputed the Sudo command and it did work but it disabled my sleep shortly after only this time when I unplug the power adapter the sleep is back to normal.

Currently my charge is at 82% and the limit is set to 80%.
Sailing is on. Disable charge when sleep is on and auto discharge is on.
Im not sure If this is normal and the sleep will be back to normal once the charge limit is reached?

MatthiasKerbl, 2022-10-23:
> Not really sure if this is fixed or its a normal thing but my sleep button is greyed out. The version is 1.19 and OS 12.6 MacBook Air (M2, 2022)
>
> The sleep button is greyed out whether or not the [...]

Hi @itzTimmyy,

When you have Disable Sleep until Charge Limit activated, sleep will be greyed out until the Charge Limit is reached. Furthermore, this works in both directions which means that if your current Charge Limit is lower than the current battery percentage and you have Discharge or Automatic Discharge activated, sleep will be greyed out until the Charge Limit is reached. The same behavior goes for the other directions, when the Charge Limit is higher than the current battery percentage, sleep will be greyed out until the Charge Limit is reached. However, when sleep is greyed out and you unplug your MacBook, sleep will always automatically be activated again but sometimes it can take up to 30 seconds.

Hope that helps.

Cheers,
Matthias

kreimben, 2022-11-30:
> I have found that uninstalling and reinstalling the the `Helper Application` also works, but I'm not yet sure for how long.

It works! Thanks!

xin4869, 2025-02-23:
same issue here! How to fix??

MatthiasKerbl, 2025-02-24:
> same issue here! How to fix??

Hi @xin4869,

This is the intended behavior of AlDente. You can read more about it on our blog ["When Does AlDente Disable Sleep on MacBooks?"](https://apphousekitchen.com/when-does-aldente-disable-sleep-on-macbooks/)

Best,
Matthias