AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #563: What is the priority support email?

jasonesiegel, 2022-01-21:
I bought a license but did not save the email address

MatthiasKerbl, 2022-01-22:
Hi @jasonesiegel,

You can write us at news@apphousekitchen.com

Cheers,
Matthias

closed 2022-01-22 by MatthiasKerbl

Omairfp, 2022-12-20:
I purchased Aldente pro but yet didn't get licence code my email ID - omairfp@gmail.com....what to do now??

MatthiasKerbl, 2022-12-22:
> I purchased Aldente pro but yet didn't get licence code my email ID - [omairfp@gmail.com](mailto:omairfp@gmail.com)....what to do now??

Hi @Omairfp,

I am sorry you are having trouble with the buying process.
Please take a look at the following steps from our FAQ:

**I Did Not Get The License Code Or Any Receipt Emails After Buying AlDente Pro!**

There are multiple reasons why you did not get your license code after the purchase:

1. Sometimes it takes up to 30 Minutes for the purchase emails to go through.
2. The purchase emails end up in the junk email folder.
3. You had a typo in the email address during your purchase.
4. Failures of the payment networks/providers.

Please contact the support team of our payment provider Paddle at [assist@paddle.com](mailto:assist@paddle.com) and include the email address you used to purchase AlDente Pro. The Paddle support will try to solve your issue as fast as possible.

Cheers,
Matthias