AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #564: AlDente 1.13 doesn't seem to respect 80% charging limit on m1 max when usb c power is plugged in

arooni, 2022-01-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
[image]

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Self explanatory; it appears as though al dente is allowing charge to go above 80% even when its requested to stay at 80 or below

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - MacBook Pro (16-inch, 2021)
 - macOS version (https://support.apple.com/en-us/HT201260)
 - 12.1
 - AlDente Pro or AlDente Free?
 - Free
 - AlDente Version?
 - 1.13
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Add any other context about the problem here.

arooni, 2022-01-22:
The power source is a 25W USB C Samsung PD charger => usb c cable =>  mac book pro

GvXV3toYUNbmP, 2022-01-22:
- did you click discharge button ?
- MacSafe is working fine ?

Somberland, 2022-01-22:
Got the same Problem with Magsafe, always charges to 100%, even tried the calibration-mode. had to unplug otherwise it won't discharge :(

GvXV3toYUNbmP, 2022-01-22:
could be same issue as https://github.com/davidwernhart/AlDente/issues/534
I am using old version 1.08 and it is working fine. 1.12 was working fine too

skynet01, 2022-01-27:
Same here, on M1 pro 16. The older 1.12 was working fine too.

MatthiasKerbl, 2022-01-29:
> **Describe the bug** A clear and concise description of what the bug is.
>
> **To Reproduce** Steps to reproduce the behavior: [image]
>
> 1. Go to '...'
> 2. Click on '....'
> 3. Scroll down to '....'
> 4. See [...]

Hi @arooni,

I can see from your screenshot that charging is stopped at 84% at the moment. In which situation does the charge limiter not work?

Cheers,
Matthias

closed 2022-02-25 by MatthiasKerbl