AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #567: Not working when 100% (Aldente Free, M1 Air, Monterey)

lyukidon, 2022-01-25:
**Describe the bug**

Before 100%( battery <= 99%), aldente works well, but after fully charged (100%), it does not work at all. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.
[image]

**MacBook (please complete the following information):**
 - MacBook model
M1 Air (a2337)
 - macOS version
12.1 Monterey
 - AlDente Pro or AlDente Free?
 Aldente Free
 - AlDente Version?
 1.13
 - List of all enabled/disabled settings in AlDente:
60% limit

**Additional context**
Add any other context about the problem here.

MatthiasKerbl, 2022-01-25:
Hi @lyukidon,

What exactly do you mean by "Not working when 100%"?
AlDente is meant to limit your charging to less than 100%

Cheers,
Matthias

lyukidon, 2022-01-25:
Thank You for Reply, sir.

When Macbook turned off while plugged in, it was fully charged. So... After
I used my mac, I wanted just plug-In mode(? not charging mode), so I
thought that It didn't work.

Sorry for not good at English, and Have a good Day!! :)

MatthiasKerbl, 2022-01-26:
Hi @lyukidon,

No worries. :)
You can read more about how AlDente handles powered-off MacBooks and Sleep Mode in our blog:
https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/
https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/

However, these explanations are for AlDente Pro.

Cheers,
Matthias

closed 2022-01-26 by MatthiasKerbl