AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #572: Automatic calibration

abidk, 2022-01-27:
**Is your feature request related to a problem? Please describe.**
The recommendation is to run the calibration once in a while to reset the battery.

**Describe the solution you'd like**
It would be good if AlDente did an automatic calibration, this would be much easier than the user manually doing it.

retrospct, 2022-01-28:
@abidk I feel like Automatic Calibration combined with my feature request in #575 will be a welcomed addition.

MatthiasKerbl, 2022-01-29:
Hi @abidk,

Good idea, we might combine this with another new feature.

Cheers,
Matthias

avxkim, 2022-02-05:
Would be cool to see this in app. Just thought about this today.

MatthiasKerbl, 2023-03-12:
Hi everyone,

We have just released AlDente 1.21.1 which has a new beta feature called Schedule. This feature allows users to set specific Actions in AlDente to run automatically on predetermined days, times, and intervals. For example, Calibration Mode can be set to run every two weeks.
You can read more about the Schedule feature and its current limitations here: https://apphousekitchen.com/feature-explanation-schedule/
You can find a full list of the Release Notes here: https://apphousekitchen.com/changelog/

Cheers,
Matthias

MatthiasKerbl, 2023-06-09:
Hi everyone,

Since our Schedule feature has officially been released for a while now, I will close this issue/feature request.

Best,
Matthias

closed 2023-06-09 by MatthiasKerbl