AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #573: Screen stays on while lid is closed

layandreas, 2022-01-27:
**Describe the bug**
The screen stays on even when lid is closed.

**MacBook (please complete the following information):**
 - Macbook Air M1 2020
 - macOS Monterey
 - AlDente Free
 - AlDente 1.13

**Additional context**
It may be related to the sleep / discharging while sleeping issues reported here.

HayriGomer, 2022-01-28:
16 M1 Pro have the same issue. Restarting the app fixes it. 
Temp solution: turn off run at start up and launch it manually

rafajrichard, 2022-01-29:
Can confirm, M1 Pro 14" does not go to sleep, exiting the app fixes it

MatthiasKerbl, 2022-01-29:
> **Describe the bug** The screen stays on even when lid is closed. Problem is solved after exiting AlDente.
> 
> **MacBook (please complete the following information):**
> 
> * Macbook Air M1 2020
> * macOS Monterey 12.2
> * AlDente Free
> * AlDente 1.13
> 
> **Additional context** It may be related to the sleep / discharging while sleeping issues reported here.
> 
> **Screenshots** [image]
> 
> [image]

Hi @layandreas,

Thanks for reporting, I am currently testing a new beta version with a potential fix for this issue.
It would still be helpful if you could share a debug file of AlDente when this happens. https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/

Cheers,
Matthias

layandreas, 2022-01-30:
Hi @MatthiasKerbl,
sure I can do that if you provide me with a mail address to send it to (I can't attach it here unfortunately).

MatthiasKerbl, 2022-01-31:
> Hi @MatthiasKerbl, sure I can do that if you provide me with a mail address to send it to (I can't attach it here unfortunately).

You can send it to news@apphousekitchen.com

Alraku, 2022-01-31:
Hi, MacBook Pro 13" 2020 M1 same problem. Yesterday I was watching Netflix and before going sleep I checked the battery status - 75%. Today after like 12hours I received an email sound from my laptop - which was strange cause it should be in sleep mode. It occurred that display was on all night and the battery dropped to around 50%.

MatthiasKerbl, 2022-02-02:
Hi There,

We have just released a new version of AlDente (1.14) which should fix the issues you experienced.
If you still encounter any of the issues after updating, please report back.

Download AlDente 1.14 here: https://github.com/davidwernhart/AlDente/releases

Cheers,
Matthias

closed 2022-02-02 by MatthiasKerbl