AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #574: AlDente will cause a crash on the 12.2 system?

kentzh1, 2022-01-28:
The day before yesterday, I got my exchanged 16-inch M1 max with 64G memory, and updated the system version to 12.2, and the AlDente Pro version to 1.13. I left after starting the screen saver in the past two days. After I came back, the computer automatically restarted or shut down. When I check the console crash report, the process name has AlDente. I am not sure whether it is AlDente that conflicts with the 12.2 version. I will upload the crash report. I hope the developer can troubleshoot this problem. Whether it is an AlDente problem
[LOG.zip](https://github.com/davidwernhart/AlDente/files/7958870/LOG.zip)
.

MatthiasKerbl, 2022-01-30:
Hi @kentzh1,

We looked into your crash report and it seems like when macOS restarted for the first time after the update, there was some kind of an anomaly when the AlDente logo got loaded. However, if it does not occur anymore, it was probably only a one-time thing. If you encounter this issue again, please let us know.

Cheers,
Matthias

closed 2022-01-30 by MatthiasKerbl

DamonHD, 2022-01-30:
FWIW, in case it is related, fully waking up Mac up from sleep and getting to the password prompt when AlDente is running seems now to take ~30s rather than a couple of seconds, so maybe there is something afoot there?

Rgds

Damon

kentzh1, 2022-02-02:
Damon

Just panicked and shut down again, please help to read the crash report, whether it is caused by software.

panic(cpu 2 caller 0xfffffe001ddb26bc): Sleep transition timed out after 35 seconds while calling power state change callbacks. Suspected bundle: com.apple.driver.AppleTypeCPhy. Thread 0x40c6c.
Debugger message: panic
Memory ID: 0x6
OS release type: User
OS version: 21D49
Kernel version: Darwin Kernel Version 21.3.0: Wed Jan  5 21:37:58 PST 2022; root:xnu-8019.80.24~20/RELEASE_ARM64_T6000
Fileset Kernelcache UUID: 6FE5ED520EAABAFE218814C92CA517B4
Kernel UUID: BADF56F4-2876-3FF4-AC12-F25E78B09AA1
iBoot version: iBoot-7429.81.3
secure boot?: YES
Paniclog version: 13
KernelCache slide: 0x0000000015c74000
KernelCache base:  0xfffffe001cc78000
Kernel slide:      0x00000000164a8000
Kernel text base:  0xfffffe001d4ac000
Kernel text exec slide: 0x0000000016590000
Kernel text exec base:  0xfffffe001d594000
mach_absolute_time: 0x91ac5fd3ab
Epoch Time:        sec       usec
  Boot    : 0x61f9d12b 0x000ee7dd
  Sleep   : 0x61fa3110 0x000b935b
  Wake    : 0x61fa3117 0x0009ead8
  Calendar: 0x61fa3fca 0x00043867

Zone info:
Foreign   : 0xfffffe00262e0000 - 0xfffffe00262f4000
Native    : 0xfffffe10002f8000 - 0xfffffe30002f8000
Readonly  : 0xfffffe14ccfc4000 - 0xfffffe1666958000
Metadata  : 0xfffffe7d1327c000 - 0xfffffe7d1f1e8000
Bitmaps   : 0xfffffe7d1f1e8000 - 0xfffffe7d36bc8000
CORE 0 PVH locks held: None
CORE 1 PVH locks held: None
CORE 2 PVH locks held: None
CORE 3 PVH locks held: None
CORE 4 PVH locks held: None
CORE 5 PVH locks held: None
CORE 6 PVH locks held: None
CORE 7 PVH locks held: None
CORE 8 PVH locks held: None
CORE 9 PVH locks held: None
CORE 0: PC=0xfffffe001d61af50, LR=0xfffffe001d61af4c, FP=0xfffffe6115713ef0
CORE 1: PC=0xfffffe001d61af4c, LR=0xfffffe001d61af4c, FP=0xfffffe6037eb3ef0
CORE 2 is the one that panicked. Check the full backtrace for details.
CORE 3: PC=0xfffffe001d61af50, LR=0xfffffe001d61af4c, FP=0xfffffe61157c3ef0
CORE 4: PC=0xfffffe001d61af50, LR=0xfffffe001d61af4c, FP=0xfffffe6101b63ef0
CORE 5: PC=0xfffffe001d61af50, LR=0xfffffe001d61af4c, FP=0xfffffe610143bef0
CORE 6: PC=0xfffffe001d61af4c, LR=0xfffffe001d61af4c, FP=0xfffffe6037813ef0
CORE 7: PC=0xfffffe001d61af4c, LR=0xfffffe001d61af4c, FP=0xfffffe610150bef0
CORE 8: PC=0xfffffe001d61af4c, LR=0xfffffe001d61af4c, FP=0xfffffe6115703ef0
CORE 9: PC=0xfffffe001d61af4c, LR=0xfffffe001d61af4c, FP=0xfffffe61156d3ef0
Panicked task 0xfffffe2999bd8678: 0 pages, 634 threads: pid 0: kernel_task
Panicked thread: 0xfffffe16679d8000, backtrace: 0xfffffe6037bb33e0, tid: 265309
		  lr: 0xfffffe001d5e64e8  fp: 0xfffffe6037bb3450
		  lr: 0xfffffe001d5e61b8  fp: 0xfffffe6037bb34c0
		  lr: 0xfffffe001d7262bc  fp: 0xfffffe6037bb34e0
		  lr: 0xfffffe001d7186c8  fp: 0xfffffe6037bb3560
		  lr: 0xfffffe001d716118  fp: 0xfffffe6037bb3620
		  lr: 0xfffffe001d59b7f8  fp: 0xfffffe6037bb3630
		  lr: 0xfffffe001d5e5e2c  fp: 0xfffffe6037bb39d0
		  lr: 0xfffffe001d5e5e2c  fp: 0xfffffe6037bb3a40
		  lr: 0xfffffe001ddf97a0  fp: 0xfffffe6037bb3a60
		  lr: 0xfffffe001ddb26bc  fp: 0xfffffe6037bb3d80
		  lr: 0xfffffe001dcfa748  fp: 0xfffffe6037bb3db0
		  lr: 0xfffffe001d637ab4  fp: 0xfffffe6037bb3e30
		  lr: 0xfffffe001d638b08  fp: 0xfffffe6037bb3f20
		  lr: 0xfffffe001d5a4e78  fp: 0x0000000000000000
[... 234 lines cut ...]
com.apple.iokit.IOCECFamily	1
com.apple.iokit.IOAudio2Family	1.0
com.apple.driver.AppleIISController	100.1
com.apple.driver.AppleAudioClockLibs	100.9.1
com.apple.driver.AppleM2ScalerCSCDriver	265.0.0
com.apple.iokit.IOSurface	302.11.1
com.apple.driver.IODARTFamily	1
com.apple.security.quarantine	4
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.security.AppleImage4	4.2.0
com.apple.kext.CoreTrust	1
com.apple.iokit.IOCryptoAcceleratorFamily	1.0.1
com.apple.driver.AppleARMPlatform	1.0.2
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOSlowAdaptiveClockingFamily	1.0.0
com.apple.iokit.IOReportFamily	47
com.apple.kec.pthread	1
com.apple.kec.Libm	1
com.apple.kec.corecrypto	12.0

** Stackshot Succeeded ** Bytes Traced 402605 (Uncompressed 1009264) **

开 2022年1月30日

MatthiasKerbl, 2022-02-02:
Hi @kentzh1,

We have looked into your crash report and it seems like it has nothing to do with AlDente. The crash report says "Suspected bundle: com.apple.driver.AppleTypeCPhy.". Therefore, it has probably something to do with the USB-C driver.

Cheers,
Matthias

kentzh1, 2022-02-11:
Hi AlDente

I like aldente very much. I hope it can be more perfect. There is another crash today. I send the crash report to you. I hope it will be helpful to improve the product.

kent

Translated Report (Full Report Below)
-------------------------------------

Process:               AlDente [2687]
Path:                  /Applications/AlDente.app/Contents/MacOS/AlDente
Identifier:            com.apphousekitchen.aldente-pro
Version:               1.14 (21)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-02-11 10:25:57.2892 +0800
OS Version:            macOS 12.2.1 (21D62)
Report Version:        12
Anonymous UUID:        249410B2-AF16-29D5-D05F-0DD32A8ED3DE

Time Awake Since Boot: 800 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x0000000102fae36c
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [2687]

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   AlDente                       	       0x102fae36c 0x102f7c000 + 205676
1   AlDente                       	       0x102faddfc 0x102f7c000 + 204284
2   AlDente                       	       0x102fad8f0 0x102f7c000 + 202992
3   AppKit                        	       0x192691e10 -[NSApplication(NSAppleEventHandling) _handleAEReopen:] + 328
4   AppKit                        	       0x1924eb010 -[NSApplication(NSAppleEventHandling) _handleCoreEvent:withReplyEvent:] + 780
5   Foundation                    	       0x1908367c4 -[NSAppleEventManager dispatchRawAppleEvent:withRawReply:handlerRefCon:] + 344
6   Foundation                    	       0x190836600 _NSAppleEventManagerGenericHandler + 96
7   AE                            	       0x19612b75c 0x19611f000 + 51036
8   AE                            	       0x19612b0b4 0x19611f000 + 49332
9   AE                            	       0x196123f84 aeProcessAppleEvent + 488
10  HIToolbox                     	       0x1985383b0 AEProcessAppleEvent + 68
11  AppKit                        	       0x1924e53ec _DPSNextEvent + 1848
12  AppKit                        	       0x1924e38a4 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1332
13  AppKit                        	       0x1924d59b4 -[NSApplication run] + 596
14  AppKit                        	       0x1924a7088 NSApplicationMain + 1064
15  AlDente                       	       0x102f83668 0x102f7c000 + 30312
16  dyld                          	       0x1031f90f4 start + 520

Thread 1:
0   libsystem_pthread.dylib       	       0x18f8c0010 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	       0x18f8c0010 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	       0x18f8c0010 start_wqthread + 0

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x000060000142c9a0   x1: 0x00000001d9e3a93e   x2: 0x0000000000000001   x3: 0x000000016ce82b91
    x4: 0x00000001e8f13f10   x5: 0x0000000103256740   x6: 0x0000000000000001   x7: 0x0000000000000000
    x8: 0x0000000040000000   x9: 0x0000000000000001  x10: 0x0000000000000000  x11: 0x000060000142c9b0
   x12: 0x0000000000000001  x13: 0x000000014a835fd0  x14: 0x0000000100000000  x15: 0x00000001e898c978
   x16: 0x00000001e898c978  x17: 0x855d8001924af5c8  x18: 0x0000000000000000  x19: 0x0000600000d24e00
   x20: 0x000060000142c9a0  x21: 0x0000000000000000  x22: 0x0000000000000000  x23: 0x0000600002337a20
[... 228 lines cut ...]
      "rolloutId" : "5fc94383418129005b4e9ae0",
      "factorPackIds" : {

      },
      "deploymentId" : 240000231
    },
    {
      "rolloutId" : "5ffde50ce2aacd000d47a95f",
      "factorPackIds" : {

      },
      "deploymentId" : 240000106
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "c4ac6461-f0e0-47b0-a630-f7739342ca3b",
      "experimentId" : "60c785f07aa5af21f0101872",
      "deploymentId" : 400000012
    }
  ]
}
}

开 2022年2月2日

MatthiasKerbl, 2022-02-11:
> Hi AlDente I like aldente very much. I hope it can be more perfect. There is another crash today. I send the crash report to you. I hope it will be helpful to improve the product. kent Translated [...]

Hi @kentzh1,

It looks like there is some kind of bug related to the menu bar icon. Do you have the "Hide Menubar Icon" setting in Settings/Appearance enabled or disabled?

Cheers,
Matthias