AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #575: [Feature Request] Show timestamp of last successful calibration

retrospct, 2022-01-28:
As a user I'd like to know when the last successful calibration cycle happened.

I actually feel like this update with the `Automatic Calibration` feature request #572 will make a huge improvement. Thanks!

I made some quick mockups to display the idea:
1. Display time since last successful calibration.
2. Display time since in the menubar UI.
3. Move the settings sprocket button down to balance the UI from adding more space for the time since text.
[image]
[image]

MatthiasKerbl, 2022-01-29:
Hi @retrospct,

That’s a good idea and thank you for the mockups.
We will try to implement some kind of timer since the last calibration.

Cheers,
Matthias

MatthiasKerbl, 2023-03-12:
Hi @retrospct,

Unfortunately, not a timer yet but our newest version AlDente 1.21.1 has a new beta feature called Schedule. This feature allows users to set specific Actions in AlDente to run automatically on predetermined days, times, and intervals. For example, Calibration Mode can be set to run every two weeks.
You can read more about the Schedule feature and its current limitations here: https://apphousekitchen.com/feature-explanation-schedule/
You can find a full list of the Release Notes here: https://apphousekitchen.com/changelog/

Cheers,
Matthias

MatthiasKerbl, 2024-09-03:
Hi @retrospct,

We have just released AlDente 1.28.1 which now shows the last successful calibration in the settings window. To get AlDente 1.28.1 you just need to update within the app under Settings/General/Check for Updates.
Please report back if it works for you.

Best,
Matthias

closed 2024-09-03 by MatthiasKerbl