AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #576: Ability to alter current output from USB C ports on MBP

lastresort1, 2022-01-30:
I'm not sure if this can fit into the scope of your product, but I would love to alter the current output from the USB C ports.
I imagine being able to ideally alter the current output based on the sound so that the light connected to my Macbook Pro becomes dim/bright according to the music. I know it's a tall order, I just want to know if it's possible, and any way I can help.

andrewcrook, 2022-03-19:
I think its out of scope for this software. 

However, use a smart dimmer and a home automation solution which is open source and script that.
There are also some microcontroller projects out there that could be modified. 

https://learn.adafruit.com/macropad-remote-procedure-calls-over-usb-to-control-home-assistant/home-assistant-configuration

lastresort1, 2022-03-21:
Thank you for your time!