AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #577: Discharge isn't working

lord-lucario, 2022-02-02:
Discharge Feature is not working on any version of AlDente, it show that it is reached its limit (current percentage) then shows that it isn't charging (like discharge should) then shows that it's charging and repeats this whole cycle and slowly charges my Mac.

It should not charge at all with the charge plugged in and with discharge mode on. (optimised battery charging is off.)

 - MacBook model Air M1 2020
 - macOS version Montery 12.0.1
 - AlDente Free
 - All
 - List of all enabled/disabled settings in AlDente:
[image]

It was working fine until a few days ago

MatthiasKerbl, 2022-02-02:
> Discharge Feature is not working on any version of AlDente, it show that it is reached its limit (current percentage) then shows that it isn't charging (like discharge should) then shows that it's [...]

Hi @lord-lucario,

Please try our newest version AlDente 1.13.
Also, Discharge only works when the Charge Limit is lower than your current SoC and Discharge does not work at all in clamshell mode (closed-display with external monitor) when you only have the free version of AlDente.
Please try the following:

1. Uninstall your current version of AlDente.
2. Install the newest version of AlDente. https://github.com/davidwernhart/AlDente/releases/download/1.13/AlDente.dmg
3. Set the Charge Limit lower than the current battery percentage.
4. Start Discharge

Does ist work?

Cheers,
Matthias

lord-lucario, 2022-02-02:
No It doesn't, It fluctuates from charging, to not charging, and back again and I don't use clamshell mode, I use an open laptop and a monitor

MatthiasKerbl, 2022-02-02:
> No It doesn't, It fluctuates from charging, to not charging, and back again and I don't use clamshell mode, I use an open laptop and a monitor
> 
> Thanks, lord-lucario

Did you try it with our newest version AlDente 1.14?

lord-lucario, 2022-02-02:
Yes I did but to no avail, this is an identical problem as in #579 except for my hardware

lord-lucario, 2022-02-04:
Ummmm update it just fixed its self?

MatthiasKerbl, 2022-02-04:
> Ummmm update it just fixed its self?

Hi @lord-lucario,

Do you mean the bug is now gone for you with the newest version?

lord-lucario, 2022-02-04:
no I didn't download any new versions, I just tried it today and Discharge, and charge limiter worked!

closed 2022-02-04 by lord-lucario

MatthiasKerbl, 2022-02-04:
> no I didn't download any new versions, I just tried it today and Discharge, and charge limiter worked!

Thanks for reporting back and I am glad it works now.

JardiinWD, 2023-04-24:
Hi!
I notice the lack of discharge mode on my macbook as well.
Version of Al Dente : Pro **1.21.4**
Macbook operating system : macOs **Ventura 13.3.1**

In brief : I tried the free version of Al Dente free (and had discharge mode available). I liked it and decided to buy the licence of Al Dente Pro. **I uninstalled the free version + the assistant**, downloaded Al Dente Pro.dmg and reinstalled by associating my licence key.

I look forward to your feedback,
Thank you!

PhilH-dev, 2023-10-06:
We have the latest AlDente Free on a new M2 Pro with Ventura that keeps losing the will to discharge,
unlike on its predecessor, an intel with Monterey where AlDente performs as expected.