AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #579: Discharge and Charge Limit Not Working on M1 Pro MBP 16

15boyyu, 2022-02-02:
MacOS 12.1
MacBook Pro 16 M1 Pro
AlDente 1.14 Free

Charge Limiter seems to do nothing. 

When I click on the discharge button, the power source changes to battery for about 1 second, then it shifts to power adapter, after a short period, it ignores AlDente and starts to charge. 

So despite enabled discharge, it keeps cycling through the three states as the images show below, and the battery state of charge keeps going up.
[image]
[image]
[image]

lord-lucario, 2022-02-02:
I have the same problem and am seeking an answer in #577

peterfan727, 2022-02-07:
I have the same spec machine, and the Free version works perfectly fine for mine. I suggest reinstalling the Helper under General Settings and then restart your Mac.

MatthiasKerbl, 2022-02-25:
Hi @15boyyu,

Do you have Optimized battery charging in System Preferences disabled?
Did you try a complete reinstall of AlDente?

Cheers,
Matthias

closed 2022-05-11 by MatthiasKerbl

quitemadmax, 2025-06-17:
Thanks for the quick reply.
I installed the recently released version 1.33.2 and it works fine.

I'm working on translating the app into Turkmen language to get a discount on the Pro version (once I read a post about it on your website).

I really like your app! I should have started using it earlier (before its health dropped to 89%), but whenever I started, it was just the right time. 
I recommend your app to all Mac users I know.

I've also just added the link to an update on [another forum](https://forums.macrumors.com/threads/al-dente-not-working.2432008/post-33964501) I found. 
Since the problem is new, it hasn't shown up almost anywhere yet, perhaps another wanderer will find my post and will fix the problem.