AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #580: the battery is charging over the charge limit.

bincento37, 2022-02-02:
**Describe the bug**
A clear and concise description of what
[image]
 the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or screen recordings to help explain your problem.

**MacBook (please complete the following information):**
 - MacBook model (https://support.apple.com/en-us/HT201608)
 - macOS version (https://support.apple.com/en-us/HT201260)
 - AlDente Pro or AlDente Free?
 - AlDente Version?
 - List of all enabled/disabled settings in AlDente:

**Additional context**
Add any other context about the problem here.

ak-me, 2022-02-03:
I am having the same issue.  I have set it up at 70% and was 85% when I plugged my laptop in via USB-C for using external monitor.  Now it is charged to 100% as the USB-C pushes power from the dock.  
Was planning on buying Pro but this severely limited Free version does not work which makes me wonder if buying a pro will help.

My system:  MacBook Pro 16in M1 Pro Max

jeanqui, 2022-02-04:
Same for me here, its been a problem since version 1.13 and it is still present in 1.14. I think the only way to resolve it is to go back to 1.12

Sullivan-Patrick, 2022-02-05:
Same. Setting AlDente to 80% seems to limit my charge on my 2020 M1 air to 89%..

jeanqui, 2022-02-05:
Okay I just did a clean reinstall and now its working properly. I uninstalled it fully, checking the relevant Library folders, then installed the latest version (1.14) and its working.

latestnight, 2022-02-07:
I just installed AlDente 1.14 last night, and I am having the same problem on an M1 Max. First time user of AlDente - will try reinstalling. Question though - when you have your max charge limit set, do you have to do anything with the discharge/top up buttons within the app?

MatthiasKerbl, 2022-02-25:
> I just installed AlDente 1.14 last night, and I am having the same problem on an M1 Max. First time user of AlDente - will try reinstalling. Question though - when you have your max charge limit set, do you have to do anything with the discharge/top up buttons within the app?

Hi @latestnight,

Take a look at our blog for explanations about the feature of AlDente: https://apphousekitchen.com/category/featureexplanation/

Cheers,
Matthias

MatthiasKerbl, 2022-02-25:
> The battery is charging over the set limit for macbook pro M1 Max. The computer isn't on sleep mode or anything. I'm constantly using the computer and I see it go above 70 when I'm using it

Hi @jeanqui,

Are you still having this issue? Could you send us a debug file?

Cheers,
Matthias

aldente42, 2022-02-28:
Hi, just got Pro and just want to make sure …

My battery was already charged to 100% before installing.

I turned on Sailing, and the three options at the bottom, and turned off the Mac battery optimization.

However, a day later, the battery is still at 100%.

Shouldn’t it be discharging to 80%?

MatthiasKerbl, 2022-03-01:
> Hi, just got Pro and just want to make sure …
> 
> My battery was already charged to 100% before installing.
> 
> I turned on Sailing, and the three options at the bottom, and turned off the Mac battery optimization.
> 
> However, a day later, the battery is still at 100%.
> 
> Shouldn’t it be discharging to 80%?

Hi @aldente42,

Your described behavior of AlDente Pro is completely normal. The Sailing Mode will never switch to battery power. You can read more about what the Sailing Mode actually does in our blog post here: https://apphousekitchen.com/feature-explanation-sailing-mode/

If you want your battery to go down to 80%, you have to activate the Discharge feature. (https://apphousekitchen.com/feature-explanation-discharge/)

Cheers,
Matthias

closed 2022-03-02 by MatthiasKerbl