AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #582: Calibration being reset if mac goes to sleep

avxkim, 2022-02-04:
**Describe the bug**
When i start the calibration mode, then my mac goes to sleep, calibration mode is cancelled. Is there a way to preserve this mode even after a sleep? Because it's impossible to discharge from 100% to 15% without putting a mac to sleep.

**To Reproduce**
Steps to reproduce the behavior:
1. Press **Start Calibration**
2. Put your Mac to sleep
3. Resume from sleep
4. Calibration mode is cancelled

**Screenshots**

Settings:

[image]

**MacBook (please complete the following information):**
 - MacBookPro18,1
 - Monterey 12.2
 - AlDente Pro
 - AlDente Version 1.14
 - List of all enabled/disabled settings in AlDente:
 - On the screenshot above

MatthiasKerbl, 2022-02-06:
Hi @avxkim,

Thank you for reporting this bug.
We will try to fix it with the next update.

Cheers,
Matthias

MatthiasKerbl, 2022-03-03:
Hi @avxkim,

We have just released AlDente 1.15 which fixed the issue you mentioned.
You can just do the update in AlDente or download the newest version here: https://github.com/davidwernhart/AlDente/releases

Cheers,
Matthias

closed 2022-03-03 by MatthiasKerbl

avxkim, 2022-03-04:
@MatthiasKerbl thanks, you're the man! 👍