AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #585: Added progress bar for calibration mode

kentzh1, 2022-02-06:
I don't know if it is difficult to develop, because during the charging and discharging calibration process, everyone will not always pay attention to the progress of the whole process, and it may cause confusion before it is completed; my suggestion is whether to add a progress bar in the calibration mode, In addition to being beautiful, it is also very intuitive to see the progress of the current battery calibration.
[image]

stat11, 2022-02-11:
this is good advice！

ovq, 2022-04-18:
adding to this feature set:
- show in calibration settings when the last fully executed calibration happened.
- ability to set a reminder to calibrate - maybe even taking "time connected to power" into account

closed 2024-04-12 by MatthiasKerbl