AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #587: Paused/On Hold vs Not Charging/Charging

misiektoja, 2022-02-07:
Hi there,

I noticed one thing. I have MBP 14 M1 Pro and when I use built-in native optimised battery charging feature, once it drains the battery to 80% it shows **Paused** in the iStat Menus battery info and **Charging On Hold** in the battery info in system preferences.

However when I use Aldente Pro (1.14) it shows **Not Charging** in the iStat Menus battery info and **Charging (!!!)** in the battery info in system preferences.  

Are there 2 different status values possible? Shouldn't Aldente pause charging instead of not charging (or charging, lol). 

Not sure if there is a difference, but it seems Apple differentiates it somehow and I wonder if Aldente does it in a proper way. Any comments ?

Apple way:

[image]
[image]

Aldente way:

[image]
[image]

misiektoja, 2022-02-07:
I also checked what CLI cmds say and it seems the battery info in system preferences says sth wrong (as it implies it is Charging):

_**% pmset -g batt**
Now drawing from 'AC Power'
 -InternalBattery-0 (id=XXXXXX)	80%; AC attached; **not charging present: true**

 **% system_profiler SPPowerDataType**           
Power:

    Battery Information:

      Model Information:
          Device Name: XXXXXX
          Pack Lot Code: 0
          PCB Lot Code: 0
          Firmware Version: 0b00
          Hardware Revision: 100
          Cell Revision: XXXX
      Charge Information:
          The battery’s charge is below the critical level.: No
          Fully Charged: No
          Charging: No
          State of Charge (%): 80
      Health Information:
          Cycle Count: 7
          Condition: Normal
          Maximum Capacity: 100%
...............
    AC Charger Information:

      Connected: Yes
      ID: 0x7002
      Wattage (W): 94
      Family: XXXXXXXXX
      Serial Number: XXXXXXXXXXXXXXXXXXX
      Name: 96W USB-C Power Adapter
      Manufacturer: Apple Inc.
      Hardware Version: 1,0
      Firmware Version: XXXXXXX
      Charging: No_

kentzh1, 2022-02-08:
My battery info also tells me that I'm charging and that the Aldente is paused, which can have the opposite effect if there isn't a good pause to charge and keep stressing the battery.

Carlit0, 2022-02-18:
same issue on my MBP M1 Pro 14"

MatthiasKerbl, 2022-03-02:
Hi @misiektoja,

Don’t worry, when you pause charging with Aldente, your MacBook battery definitely does not charge anymore. You can see that when you click on the macOS battery icon, it says "Battery is Not Charging". Furthermore, I can see that the power consumption is much lower when charging is paused and I can see that internally no power is going into the battery anymore and the battery percentage is also not increasing. Lastly, you can even see on your screenshots when you check the CLI cmds where it says "Charging: No".

In my opinion, the battery System Preferences in macOS gets confused and thinks it is still charging since macOS itself did not pause charging. Even though when you check differently, macOS recognizes that the battery does not charge right now.

Cheers,
Matthias

closed 2022-03-02 by MatthiasKerbl