AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #596: paid for AlDente pro, but did not receive license code

houyubirdman, 2022-02-09:
i bought AlDente pro and paid with Alipay.but i still don't receive the licence now. are you cheating??

sylerly, 2022-02-09:
I have the same problem ,please handle asap

MatthiasKerbl, 2022-02-09:
Hi there,

I am sorry for the inconvenience.
There are multiple reasons why you did not get your license code after the purchase:

1. Sometimes it takes up to 30 Minutes for the purchase emails to go through.
2. The purchase emails end up in the junk email folder.
3. You had a typo in the email address during your purchase.
4. Failures of the payment networks/providers.

Please send me the email address you used to purchase AlDente Pro to news@apphousekitchen.com. Afterward, I will be able to look into what went wrong and will forward your license code as fast as possible.

Cheers,
Matthias

yangfan1995, 2022-02-09:
> I have the same problem ,please handle asap

Yes, I've met the question as you too. -_-!

MatthiasKerbl, 2022-02-09:
Hi there,
​
​Good news: Our payment provider Paddle could fix the technical issues with Alipay and everything should work as expected again.
​Unfortunately, they could only manage to refund all the money. You will get your money back in the next couple of workdays.
​Therefore, if you still want AlDente Pro, you need to buy AlDente Pro again and this time you will get two purchasing emails with the license code.
​
Sorry for all the inconvenience!

Cheers,
Matthias

closed 2022-02-09 by MatthiasKerbl