AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #597: cannot Used in my old MacBook

manchunfai, 2022-02-09:
**Describe the bug**
lt doesn't work in my old MacBook even through I reinstall AIDente and its helper.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Menu and See error "Helpv12: not found",and the limit still not works.
2. Click on "reinstall Helper"
3. write down my accout, and click 
4. Go to Menu and still See error "Helpv12: not found",and the limit still not works.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]
[image]

**MacBook (please complete the following information):**
 - MacBook Pro 13.3 2018
 - MacOS 10.15.7
 - AlDente Free
 - 2.2

MatthiasKerbl, 2022-02-09:
Hi @manchunfai,

You can try to uninstall AlDente completely manually. Please follow the steps in this guide: https://apphousekitchen.com/how-to-uninstall-aldente-pro/

Afterward, try a fresh reinstall of AlDente.

Another option would be to update to macOS 11 Big Sur or later and use the newest version of AlDente.

Cheers,
Matthias

manchunfai, 2022-02-09:
你好，谢谢你的来信，明俊晖已收到你的信息。

manchunfai, 2022-02-09:
I followed your advices and reset my SMC of Mac, but it's pity.
I try another way, and it works.
Follow that:
Open terminal,
And Enter:sudo launchctl enable system/com.davidwernhart.Helper

Just for reference.

MatthiasKerbl, 2022-02-09:
Thanks for sharing, and I am glad it works now!

closed 2022-02-09 by MatthiasKerbl

twiz911, 2022-07-01:
> sudo launchctl enable system/com.davidwernhart.Helper

I was stuck in Al Dente 2.2 unable to install the helper, despite following all the tricks in these GitHub issues until I ran that command! Thanks :)