AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #598: 1.14 sailing mode stopped working in 12.2 on MacBook Pro 14"

rimusz, 2022-02-09:
1.14 sailing mode stopped working in 12.2 on MacBook Pro 14"
It used to be discharging battery before now it stopped

MatthiasKerbl, 2022-02-09:
Hi @rimusz,

Sailing Mode will not discharge your MacBook. Instead, it will pause charging and stop charging until your battery has fallen back to the lower charge limit.
You can read more about how the Sailing Mode works on our blog here: https://apphousekitchen.com/feature-explanation-sailing-mode/

Cheers,
Matthias

closed 2022-02-09 by MatthiasKerbl

rimusz, 2022-02-09:
no, it got changed with MacOS 12.2 release, Andante sailing mode is not working correctly:
it will pause charging but will keep on power on, same behavior as without sailing mode

MatthiasKerbl, 2022-02-09:
Hi @rimusz,

Sailing Mode will never discharge your MacBook actively. It prevents your MacBook from charging when the battery percentage drops on its own (even if it is plugged in) until it reached the lower limit. You can read a precise description about the Sailing Mode on our blog: https://apphousekitchen.com/feature-explanation-sailing-mode/

If you want your MacBook to discharge to a certain level, while it is plugged in, you have to use the Discharge feature.

Cheers,
Matthias

MatthiasKerbl, 2022-02-09:
Interesting. We did not see any different behavior of Sailing Mode between 12.1 and 12.2. However, I hope it is working now as described in our article.
If you find anything else, let us know.

rimusz, 2022-02-09:
before it was working the way we have asked a while ago, now the Sailing mode is a useless feature :(

rimusz, 2022-02-09:
I have seen the behavior as per this tread https://github.com/davidwernhart/AlDente-Pro-Bug-Tracking/issues/8

MatthiasKerbl, 2022-02-09:
For me (AlDente Pro 1.14 and macOS 12.2), Sailing Mode works exactly as described by @SuperGHUB in this thread https://github.com/davidwernhart/AlDente-Pro-Bug-Tracking/issues/8 
What exactly does not work for you right now?

KAMIKAZEUA, 2022-02-16:
@MatthiasKerbl 
This actually a bug and I understand your explanation, yes it’s fully worked as you designed before 12.2 and for my case, just until reboot of the MacBook. But this is like randomly, so not always. I’ll try to explain.

So, I’m expecting that app will let MacBook to discharge(just while I simply using it) from 100% to 80% in any situation if sailing mode enabled. But after reboot(also maybe go sleep/wake) of MacBook, then if I open menu of this app, I’ll see that it’s showing me **Charge Limit** = **100%**, but it should be 80%.
I have enabled sailing mode with 5% and Stop Charging When Sleeping enabled, MacBook always plugged in to external power. Same MacBookPro 14 M1.

rimusz, 2022-02-16:
> @MatthiasKerbl This actually a bug and I understand your explanation, yes it’s fully worked as you designed before 12.2 and for my case, just until reboot of the MacBook. But this is like randomly, [...]

yes I see the same issue

KAMIKAZEUA, 2022-02-16:
And if it’s always plugged into charger, looks like it was charged while in Sleep, so this option is not working.. well, I’m not sure, maybe sometimes..