AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #601: AlDente preventing sleep and screen saver

Nicolas-MacBeth, 2022-02-10:
First off, I love the app. Great work, thanks! However I'm encountering a bug as of recently!

**Describe the bug**
It seems the AlDente helper is stopping my computer from going to sleep or activating the screen saver, whether on the lock screen or not. This only seems to happen when the battery has not yet reached the battery threshold set in AlDente. (Seems like "Disable sleep until charge limit" is on via a bug?). I tried completely uninstalling the app following [this](https://apphousekitchen.com/how-to-uninstall-aldente-pro/), reinstalling and just trying it with the default settings (I don't have the pro mode and so impossible for me to even put  "Disable sleep until charge limit" on). This is with the latest 1.14 release and when I quit the helper process via Activity Monitor, or the charge reaches the set limit, or the computer is not charging, the issue goes away.

**To Reproduce**
Steps to reproduce the behavior:
1. Download and install AlDente
2. Turn it on and set a limit
3. Computer does not activate screen saver or go to sleep

**Expected behavior**
The computer's screen saver activates properly and the computer goes to sleep, even when locked.

**MacBook:**
 - MacBook model: MacBook Pro (14-inch, 2021) - M1 Max
 - macOS version: 12.2 (21D49)
 - AlDente Free
 - AlDente Version: 1.14
 - List of all enabled/disabled settings in AlDente: All defaults

**Additional context**
Here's a link to someone else having the same problem, but they added their comment on a closed issue.
Originally posted by @lgberro in https://github.com/davidwernhart/AlDente/issues/472#issuecomment-1032979240

**Screenshots**
(Directly below is a very WIDE image, followed by another screenshot)
[image]
<br>
[image]

Elijas, 2022-02-11:
Mac doesn't seem to go to sleep on lid close, I had multiple times where I put laptop to backpack and it becomes scorching hot, with 0% battery. This is way more battery damaging than keeping it 100% without AlDente. Had to disable AlDente altogether

```
MacBook Pro (16-inch, 2019, i7)
macOS 12.2 (21D49)
AlDente Free 1.14, Settings default
```

MatthiasKerbl, 2022-02-11:
Hi @Nicolas-MacBeth and @Elijas,

Thanks a lot for the very detailed bug report, this helps a lot.
It would be helpful if you could generate and share a debug file with us when it happens the next time. You can find a guide for the debug file here: https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/

Is the Sleep option greyed out when you encounter this problem?
[image]

Cheers,
Matthias

Nicolas-MacBeth, 2022-02-11:
Hi @MatthiasKerbl thanks for the quick reply.

Here is the debug file you're asking for (I renamed it to `.txt` since GitHub didn't like `.pasta`). Also, the Sleep option is **not** greyed out.
[aldentepro-debug-2022-02-11-15:37:39.txt](https://github.com/davidwernhart/AlDente/files/8051515/aldentepro-debug-2022-02-11-15.37.39.txt)

Thanks,
Nic

yyberi, 2022-02-11:
This might be related to this also:

_macOS Monterey 12.2.1 Release Notes
Release notes attached to the update are as follows. Note that the release notes specifically mention Intel Macs, though some M1 Mac users also have noted the battery draining problem:

macOS Monterey 12.2.1 — Restart Required

macOS 12.2.1 provides important security updates and fixes an issue for Intel-based Mac computers that may cause the battery to drain during sleep when connected to Bluetooth peripherals._

Nicolas-MacBeth, 2022-02-11:
It's specifically AlDente stopping sleep in my case, when I kill the process behaviour goes back to normal! But yeah it could be what @Elijas is experiencing.

Nicolas-MacBeth, 2022-02-16:
Hey @MatthiasKerbl, just want to confirm you got my debug file above!

Wish there was a way to DM you so I could keep this thread cleaner

Nic

MatthiasKerbl, 2022-02-17:
> Hey @MatthiasKerbl, just want to confirm you got my debug file above!
> 
> Wish there was a way to DM you so I could keep this thread cleaner
> 
> Nic

Hi @Nicolas-MacBeth,

Yes, I got the debug file. Thanks a lot.
I am currently testing a beta version of AlDente with a fix for the bugs you mentioned. So far it looks promising. :)

DMs are only possible for AlDente Pro users.

Cheers,
Matthias

Nicolas-MacBeth, 2022-02-17:
Sounds good, thanks! @MatthiasKerbl

theonefoster, 2022-02-23:
I have the same issue. In Terminal, running `pmset -g` shows:

``` sleep                1 (sleep prevented by com.apphousekitchen.aldente-pro, sharingd, powerd)
 displaysleep         5 (display sleep prevented by com.apphousekitchen.aldente-pro)```

Laptop does not sleep when the lid is closed - even the screen stays on when closed - and the "sleep" option is greyed out in the Apple menu. 

This is a recent issue - it's not been a problem before the last couple of weeks.

M1 Macbook Air, macos 12.2.

Elijas, 2022-02-24:
I've been able to fix the greyed out Sleep button with

'sudo defaults write /Library/Preferences/SystemConfiguration/com.apple.PowerManagement SystemPowerSettings -dict SleepDisabled -bool NO'

MatthiasKerbl, 2022-02-24:
> I have the same issue. In Terminal, running `pmset -g` shows:
> 
> ```
> sleep                1 (sleep prevented by com.apphousekitchen.aldente-pro, sharingd, powerd)
> displaysleep         5 (display sleep prevented by com.apphousekitchen.aldente-pro)
> ```
> 
> Laptop does not sleep when the lid is closed - even the screen stays on when closed - and the "sleep" option is greyed out in the Apple menu. The only way to get the laptop to go to sleep is to quit AlDente - then normal sleep behaviour resumes.
> 
> This is a recent issue - it's not been a problem before the last couple of weeks.
> 
> AlDente v1.13, M1 Macbook Air, MacOS 12.2. Let me know if you need any more info.

Hi @theonefoster,

Please update to our newest version AlDente 1.14. This version has a couple of fixes for most sleep-related issues.

Cheers,
Matthias

theonefoster, 2022-02-25:
@MatthiasKerbl Thanks! 1.14 seems to have resolved my sleep issue :)

GJRobert, 2022-02-26:
By merely updating to 1.14 my sleep problem is not corrected yet. What else did you do?

MatthiasKerbl, 2022-03-01:
> By merely updating to 1.14 my sleep problem is not corrected yet. What else did you do?
> 
> **Update**: I have just done the method of @Elijas
> 
> > @theonefoster I've been able to fix the greyed out Sleep button with
> > `sudo defaults write /Library/Preferences/SystemConfiguration/com.apple.PowerManagement SystemPowerSettings -dict SleepDisabled -bool NO`
> 
> And now my `Sleep` menu item is back to normal.

Hi @GJRobert,

Which features do you have activated?
We are actually disabling sleep on purpose in certain situations when certain features are activated.

dcobas, 2022-03-01:
First of all, many thanks for this extraordinary app: it is outstanding and I love it!

I have the same problem as Nicolas-MacBeth reported. A fresh installation of AlDente 1.07 does not exhibit the issue. After upgrade to 1.14, display refuses to sleep, disabled by the AlDente helper. I attach herewith a debug file renamed to .txt:
[aldentepro-debug-2022-03-01-15:44:12.pasta.txt](https://github.com/davidwernhart/AlDente/files/8162375/aldentepro-debug-2022-03-01-15.44.12.pasta.txt).

This is the `pmset -g` report when the machine is connected to power. NB: the disabling assertion on displaysleep from AlDente helper disappears when unplugging the power cord and running on battery; and it reappears when powered back again.
```
cobas-m1:~ cobas$ pmset -g
System-wide power settings:
 SleepDisabled          0
Currently in use:
 standby              1
 Sleep On Power Button 1
 hibernatefile        /var/vm/sleepimage
 powernap             1
 networkoversleep     0
 disksleep            10
 sleep                0 (sleep prevented by powerd, com.apphousekitchen.aldente-pro)
 hibernatemode        3
 ttyskeepawake        1
 displaysleep         2 (display sleep prevented by com.apphousekitchen.aldente-pro)
 tcpkeepalive         1
 lowpowermode         0
 womp                 1
```
Machine: MacBook Air (M1, 2020)
OS: macOS Monterey 12.2.1 (21D62)
RAM: 8GB
AlDente: 1.14 Pro
Options set at defaults (see screenshots attached).

[image]
[image]

MatthiasKerbl, 2022-03-01:
> First of all, many thanks for this extraordinary app: it is outstanding and I love it!
> 
> I have the same problem as Nicolas-MacBeth reported. A fresh installation of AlDente 1.07 does not exhibit the issue. After upgrade to 1.14, display refuses to sleep, disabled by the AlDente helper. I attach herewith a debug file renamed to .txt: [aldentepro-debug-2022-03-01-15:44:12.pasta.txt](https://github.com/davidwernhart/AlDente/files/8162375/aldentepro-debug-2022-03-01-15.44.12.pasta.txt).
> 
> This is the `pmset -g` report when the machine is connected to power. NB: the disabling assertion on displaysleep from AlDente helper disappears when unplugging the power cord and running on battery; and it reappears when powered back again.
> 
> ```
> cobas-m1:~ cobas$ pmset -g
> System-wide power settings:
>  SleepDisabled          0
> Currently in use:
>  standby              1
>  Sleep On Power Button 1
>  hibernatefile        /var/vm/sleepimage
>  powernap             1
>  networkoversleep     0
>  disksleep            10
>  sleep                0 (sleep prevented by powerd, com.apphousekitchen.aldente-pro)
>  hibernatemode        3
>  ttyskeepawake        1
>  displaysleep         2 (display sleep prevented by com.apphousekitchen.aldente-pro)
>  tcpkeepalive         1
>  lowpowermode         0
>  womp                 1
> ```
> 
> Machine: MacBook Air (M1, 2020) OS: macOS Monterey 12.2.1 (21D62) RAM: 8GB AlDente: 1.14 Pro Options set at defaults (see screenshots attached).
> 
> [image]
> 
> [image]
> 
> NB: after reaching the 80% set limit, the assertion disappears and the display may go to sleep again.

Hi @dcobas,

I am glad you like our app AlDente. :)

Thank you for this detailed bug report. As mentioned above, an update will be released soon with a fix for this issue.

Cheers,
Matthias

MatthiasKerbl, 2022-03-03:
Hi everyone,

We have just released AlDente 1.15 which has a bug fix for the sleep prevention issue.
You can just do the update in AlDente or download the newest version here: https://github.com/davidwernhart/AlDente/releases

Cheers,
Matthias

closed 2022-03-03 by MatthiasKerbl

Nicolas-MacBeth, 2022-03-07:
Thanks! Seems to work on my end. The hard work and quick resolution is appreciated.

dcobas, 2022-03-08:
Same in my machine, now it works flawlessly. Thanks a lot for solving this so quickly, and for this great app!

MatthiasKerbl, 2022-03-08:
Hi @Nicolas-MacBeth and @dcobas,

I am glad it works now and thank you for the compliments. :)

Cheers,
Matthias

bwwm, 2022-07-17:
This is still not working for me. The computer does not sleep with this app running.
[image]
[image]
[image]

bwwm, 2022-07-17:
I can sleep the computer when I quit Aldente

sircoderin, 2023-11-29:
This problem is back on M1 Sonoma 14.1.1