AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #602: After restarting, the battery has changed from 80 to 77, is there a problem?

kentzh1, 2022-02-11:
*I had a battery calibration on February 7th, the computer has been trying out the AlDente pro, recently found a battery issue, I've been limited to 80 percent, but after I restarted it went to 77, which gave me Very confused, I think there is a problem with the software control, it is impossible to restart and consume 3% of the power, or is there any problem with the software, before I updated 12.2.1, this problem also appeared, when I shut down or restarted , 80% of the battery will change, hope to find out how this problem is caused.

*MacBook (please complete the following information):**
 - MacBook model (MacBook Pro2021)
 - macOS version (12.2.1)
 - AlDente Pro
 - AlDente 1.14
 - List of all enabled/disabled settings in AlDente:
 -
[image]
[image]
[image]

kentzh1, 2022-02-20:
Unfortunately, the same problem happened just now, the battery remained at 80 before the restart, and after the restart it became 77, I turned on the sailing mode, the above can not be adjusted, the difference of 5 will sail to 75 , I will close it and test again
[image]
.

kentzh1, 2022-02-22:
agai
[image]
n

MatthiasKerbl, 2022-02-24:
Hi @kentzh1,

From my experience, this is completely normal and I experience it sometimes too.

Cheers,
Matthias

closed 2022-02-24 by MatthiasKerbl