AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #603: Add Low Power Mode Button to Menu Bar Widget

ovq, 2022-02-12:
The subject says it all. It would be nice to be able to toggle low-power mode via the AlDente menu bar.

MatthiasKerbl, 2022-02-25:
Hi @ovq,

Good idea but I do not know if it is possible for third-party apps to alter these settings.
We will look into this.

Cheers,
Matthias

andrewcrook, 2022-03-19:
> but I do not know if it is possible for third-party apps to alter these settings.

I think it can be done from` pmset` 

```
arm64 andrew@Andrews-MBP ~ % pmset -g  

System-wide power settings:
 SleepDisabled		1
Currently in use:
 standby              1
 Sleep On Power Button 1
 hibernatefile        /var/vm/sleepimage
 powernap             1
 networkoversleep     0
 disksleep            10
 sleep                1 (sleep prevented by bluetoothd, com.bombich.ccchelper, com.bombich.ccchelper, powerd, sharingd, com.apphousekitchen.aldente-pro)
 hibernatemode        3
 ttyskeepawake        1
 displaysleep         10 (display sleep prevented by com.apphousekitchen.aldente-pro)
 tcpkeepalive         1
 **lowpowermode         0**
 womp                 1
```

andrewcrook, 2022-03-19:
Perhaps the feature should turn on automatically when battery life reaches a certain percentage set by the user.

basmets6, 2022-04-17:
+1 on this. I'd prefer an on-off switch, since I typically have it on at all times unless I notice my performance lagging.

andrewcrook, 2022-04-18:
Both options then

DamonHD, 2022-06-29:
And I'd like low-power mode automatically engaged when any grid-friendly features are switched on.

BTW, in a brief since test on my MBA M1, typical draw from mains adaptor with an editor prompt open and a little bit of other activity was about 4.9W dropping to about 4.2W with low-power mode on, so a useful 15%/0.7W saving in that case.

http://www.earth.org.uk/Hey-Siri-Help-Flatten-The-Duck.html#Measured

Rgds

Damon

xsnaruto, 2024-03-05:
The problem to me is I'd like to set low battery mode on when MacBook is using battery, but after aldent changed the battery as  discharging, the system will get into low battery mode, even the device still connected the power, this will cause many of function stop operating, such as iCloud photo sync etc. sometimes I forgot to manually turn sync on, I take MacBook out after a whole night with power connected, than I found out that none photos synced.

hope there's a way to let aldent turn off low battery mode after connected to the power and turn it on after detected the power is disconnected. for now, I just setup never turn on the low battery mode, because im easy to forget switch them, its not a good way to solve problem.

MatthiasKerbl, 2024-04-12:
Hi everyone,

We have just released AlDente 1.26.1 which includes a Low Power Mode setting. You can enable the Low Power Mode in Settings/Features/Low Power Mode. Also, the Right Click feature on the menubar icon is now customizable and can now be configured to trigger the Low Power Mode

To get AlDente 1.26.1 you just need to update within the app under Settings/General/Check for Updates.

Best,
Matthias

closed 2024-04-12 by MatthiasKerbl

e-kotov, 2024-04-12:
@MatthiasKerbl, love the feature! However, it's not immediately apparent whether the low power mode is activated. Currently, one must navigate to the system battery widget in the menu bar, click on it, and check the popup. Could you integrate a low power mode indicator directly into the AlDente menu bar icon or its popup with a single click? Additionally, incorporating a toggle for low power mode in the same popup would be extremely useful.

MatthiasKerbl, 2024-04-12:
> @MatthiasKerbl, love the feature! However, it's not immediately apparent whether the low power mode is activated. Currently, one must navigate to the system battery widget in the menu bar, click on [...]

Hi @e-kotov,

Thank you for your feedback and good to hear that you like the new feature.
You can see the Low Power Mode status in the AlDente popover when you have the Electrical Specs Sensor Data panel enabled in settings.
If you want to quickly enable/disable the Low Power Mode, please use the Right Click feature described above.
[image]

Best,
Matthias

colinbrislawn, 2024-07-17:
> integrate a low power mode indicator directly into the AlDente menu bar icon or its popup with a single click?

This is what I was looking for!

> Also, the Right Click feature on the menubar icon is now customizable and can now be configured to trigger the Low Power Mode

Does this mean that you can configure this so that a right-click toggels low-power mode on and off?

(This the specific UX I'm looking for, so I'm trying to understand it before buying Al Dante.)

MatthiasKerbl, 2024-07-18:
> This is what I was looking for!
>
>
> Does this mean that you can configure this so that a right-click toggels low-power mode on and off?
>
> (This the specific UX I'm looking for, so I'm trying to [...]

Hi @colinbrislawn,

Yes, you can configure AlDente Pro so that a right-click on the menu bar icon enables/disables the Low Power Mode.

Best,
Matthias

FrancisMorelli, 2024-09-29:
@MatthiasKerbl What is low power mode? I've not found an explanation on the website like others features.

e-kotov, 2024-09-29:
@FrancisMorelli it's Apple's default feature, see https://support.apple.com/en-gb/guide/mac-help/mh35848/mac

FrancisMorelli, 2024-09-29:
> @FrancisMorelli it's Apple's default feature, see https://support.apple.com/en-gb/guide/mac-help/mh35848/mac

So, it's not an AlDente feature... does it only enable the built-in Apple feature?

e-kotov, 2024-09-29:
> > @FrancisMorelli it's Apple's default feature, see https://support.apple.com/en-gb/guide/mac-help/mh35848/mac
> 
> @e-kotov  So, it's not an AlDente feature... does it only enable the built-in Apple feature?

That's what I would expect.