AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #606: i thing it not working

inclub9, 2022-02-20:
**Describe the bug**
-saling mode is not working ?
-set limit when shutdown is not working ?

**To Reproduce**
Steps to reproduce the behavior:
1. before shutdown battery 80
2. open my macos battery 100
3. set limit at 80

**Screenshots**
[image]

**MacBook (please complete the following information):**
 - MacBook model MacBook Pro (16-inch, 2021)
 - macOS version 12.2.1 (21D62)
 - AlDente Pro
 - AlDente Version? //1.14
 - List of all enabled/disabled settings in AlDente: at the picture

MatthiasKerbl, 2022-02-24:
Hi @inclub9,

What exactly does not work with the Sailing Mode? You can read everything about the Sailing Mode on our blog: https://apphousekitchen.com/feature-explanation-sailing-mode/

Regarding the behavior when you turn off your MacBook. Due to hardware changes with the new Apple Silicon MacBooks, it is not possible anymore for AlDente to pause charging even when the MacBook is shut down. That’s why the feature is called "Stop charging when app closed" and not "Stop charging when powered off". You can read everything about those features and differences for different MacBook models on our blog:
https://apphousekitchen.com/feature-explanation-stop-charging-when-powered-off-app-closed/

Cheers,
Matthias

closed 2022-02-24 by MatthiasKerbl

inclub9, 2022-02-26:
> 

ok, thanks