AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #608: macOS does not respect charge limit when switching between user accounts

samwzlim, 2022-02-22:
**Describe the bug**
MacBook charges beyond charge limit when switching between user accounts via 'fast user switching menu'. E.g. In User A, my macbook charges until the charge limit and stops, but after switching to User B, the charging resumes until it is fully charged. AlDente is up and running on both users and set to the same charge limit. Similar bug report [here](https://github.com/davidwernhart/AlDente/issues/559).

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a new user in macOS
2. Enable 'show fast user switching menu' in System Preferences > Users & Groups
3. Run AlDente on both users and set charge limit to same percentage (e.g. 80%)
4. Wait until battery is charged to set limit
5. Switch users via the fast user switching menu and notice the battery resumes charging beyond the set limit

**Expected behavior**
MacBook to stop charging at set charge limit even when switching between different user accounts.

**MacBook (please complete the following information):**
 - MacBook model: MacBook Pro 14" M1 Pro
 - macOS version: macOS 12.2.1
 - AlDente Pro or AlDente Free? AlDente Free
 - AlDente Version? v1.14
 - List of all enabled/disabled settings in AlDente: Launch at login is enabled

arthurzqhu, 2022-02-22:
Was about to post the exact same issue (for AlDente Pro).

MrBoney, 2022-02-22:
Same issue here too. Very frustrating as my normal use pattern is to switch between user accounts several times during the day. As it's managing hardware it really should function at the machine level not the user account level.

arthurzqhu, 2022-02-24:
Turns out reinstalling the helper application every time I switch user is a workaround - just a little too cumbersome.

samwzlim, 2022-02-24:
> Turns out reinstalling the helper application every time I switch user is a workaround - just a little too cumbersome.

Can confirm this works; just annoying to have to do it every time I switch between accounts :(

MrBoney, 2022-02-24:
Not a solution for me. I swap use accounts at least 10 times during the working day and having to constantly fiddle with the software to keep it working is not what I paid GBP20 for!

samwzlim, 2022-02-24:
It neither is for me. This bug did not exist in previous versions of AlDente, and I am hoping it will be solved in future releases. I do not wish to downgrade to a previous version just for this issue; at the expense of new features. However, at the moment, this seems to be the only workaround until this bug gets patched.

dylanmc, 2022-02-28:
I'm having this issue, and also see it's a duplicate of #559

MatthiasKerbl, 2022-03-01:
Hi everyone,

Thank you for reporting this issue.
We are currently working on an update that will add the fast user switching functionality.
It will be released as soon as possible.

Thank you for your patience.

Cheers,
Matthias

MatthiasKerbl, 2022-03-03:
Hi everyone,

We have just released AlDente 1.15 which finally supports Fast User Switching.
You can just do the update in AlDente or download the newest version here: https://github.com/davidwernhart/AlDente/releases

Also, please read the tutorial and explanation of how to get the Fast User Switching to work properly here: https://apphousekitchen.com/fast-user-switching-explanation-and-tutorial/

Hope you like it.

Cheers,
Matthias

closed 2022-03-03 by MatthiasKerbl

arthurzqhu, 2022-03-04:
Thanks for the update! Any idea how long it usually takes for Setapp subscription users to get the updated version?

MatthiasKerbl, 2022-03-04:
> Thanks for the update! Any idea how long it usually takes for Setapp subscription users to get the updated version?

The Setapp version is currently in review by Setapp and should be released soon.

MrBoney, 2022-03-04:
Same issue is happening with 1.15. Upgraded AlDente, rebooted and checked settings. Both users set to charge to 80%. User A charged to 80% and paused. Switched to userB and it charged to 100% despite 80% charge limit. I discharged B back to 76%, then switched back to A which charged back to 100% despite 80 limit. Exactly the same behaviour as version 1.14.

Full details: Macbook Pro 14@ M1 Pro, macOS 12.2.1, AlDente PRO 1.15. Launch at login enabled all users, sailing enabled 5% all users, heat protection disabled all users, stop when sleeping disabled all users, stop when app closed disabled all users, disable sleep until charge limit disabled all users. Charge % set to 80 all users. All settings identical for both users.

Will check back for updates after a while but deleting the app for now as it simply doesn't work for multiple user accounts.

DamonHD, 2022-03-04:
Was your AlDente slider showing as green or blue or grey while this was happening?

(For me my slider is almost always grey implying that the helper is dead or not communicating, so attempts to manage the charge level silently fail.  Though since I'm in Intel mode, the previously-set limit prevails.)

Rgds

Damon

MrBoney, 2022-03-04:
Green, blue or orange, depending on charging, paused or discharging. Never grey.

I've removed the app now though as it just doesn't work for my usage - I switch user accounts  many times a day and it just ends up unecessarily adding charge cycles to the battery (It goes to 100, I discharge back to 80, swicth users and it goes to 100 again, and so on). For now I'll just let it sit at 100 and make sure I run it down once a week.

MatthiasKerbl, 2022-03-04:
> Same issue is happening with 1.15. Upgraded AlDente, rebooted and checked settings. Both users set to charge to 80%. User A charged to 80% and paused. Switched to userB and it charged to 100% [...]

Hi @MrBoney,

I am sorry you are still encountering this problem.
We have tested the Fast User Switching on all of our test devices over the last weeks and it worked without any issues.

Could you please send us a debug file the next time this bug occurs?
https://apphousekitchen.com/how-to-generate-and-share-a-debug-file/

Cheers,
Matthias