AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #610: This shit Aldente GUI

UnixCro, 2022-02-24:
I don't find the program itself bad as it saves my Mac's battery. But what the heck. Why when I open Aldente is it in the dock & at the top of the menu bar at the same time. What if I accidentally quit it with "Force Quit". It can't be such an app can only be in the menu bar at the top. And I want it to run 24/7 and not charge my battery to 100 when it's off. Why is there no LaunchDaemon??!

MatthiasKerbl, 2022-02-24:
Hi @UnixCro,

Earlier versions of AlDente only had the little icon in the menubar. However, many new users did not find it and complained about that. That’s why we made some changes such as adding the dock icon and opening the popup menu on every start of AlDente. I completely understand that those things are annoying for most users, that’s why you can easily disable them in the appearance settings in AlDente and they will stay disabled afterward.

Hope you understand our decision.

Cheers,
Matthias

closed 2022-02-24 by MatthiasKerbl

UnixCro, 2022-02-25:
Yes, that's exactly why it's only available in the Pro version .... Who thought up this crap

MatthiasKerbl, 2022-02-25:
> Yes, that's exactly why it's only available in the Pro version .... Who thought up this crap

Hi @UnixCro,

The two options to disable the dock icon and to disable the popup on startup are available in the free version as well.

Cheers,
Matthias

UnixCro, 2022-02-25:
And where?
[image]

MatthiasKerbl, 2022-02-25:
You have to deactivate "Beim Start anzeigen" and "Dock-Symbol anzeigen".

Cheers,
Matthias

UnixCro, 2022-02-25:
LOL it worked. Sorry Bro for my Mistake.

Have a nice day