AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #615: AlDente Pro 1.14 ignored when sleeping?

tliet, 2022-03-03:
On my Intel 15" MBP from 2018 as well as my M1 MBA from 2020 AlDente Pro seems to be ignored since Mac OS 12.2.1 (21D62). 

**To Reproduce**
Steps to reproduce the behavior:
1 Battery is ~25%
2 Put Mac to sleep
3 Plug in charger
4 After a number of hours wake up Mac, the battery is fully charged (100%)

**Expected behavior**
The setting in AlDente is charge to 80%.

**Screenshots**
[image]
[image]

**MacBook (please complete the following information):**
 - MacBook Pro (15-inch, 2018) and MacBook Air (M1, 2020)
 - macOS version 12.2.1 (21D62) and macOS version 12.3 Beta (21E5222a)
 - AlDente Pro 
 - AlDente 1.14
 - List of all enabled/disabled settings in AlDente: see screenshot

MatthiasKerbl, 2022-03-03:
Hi @tliet,

Apple does not allow third-party apps to run in the background when your MacBook is asleep. That is why we had to come up with different features to deal with that.
Take a look at the features "Stop charging when sleeping" and "Disable sleep until charge limit". Please also read our explanations about those features on our blog:
https://apphousekitchen.com/feature-explanation-stop-charging-when-sleeping/
https://apphousekitchen.com/feature-explanation-disable-sleep-until-charge-limit/

Cheers,
Matthias

closed 2022-03-03 by MatthiasKerbl

tliet, 2022-03-04:
Thanks for the clarification. In my memory both Macs changed behaviour, but apparently this was incorrect.